verilator/verilator #5906: Segmentation fault

srcejon, 2025-03-31:
I get a Segmentation fault when running the attached example:

        verilator --binary --top-module uart_tb uart.v fifo.v uart_tb.v
        ./obj_dir/Vuart_tb
        idx           0
        idx           1
        Segmentation fault (core dumped)
        
Code runs okay with iverilog:
  
    iverilog -g2012 uart_tb.v uart.v  fifo.v
    ./a.out
    idx           0
    idx           1
    idx           2
    idx           3
    idx           4
    idx           5
    idx           6
    idx           7
    idx           8
    idx           9
    idx          10
    idx          11
    idx          12
    idx          13
    idx          14
    uart_tb.v:219: $finish called at 5225251000 (1ps)
    
> What 'verilator --version' are you using?  Did you try it with the git master version?

Verilator 5.020 2024-01-01 rev (Debian 5.020-1)
Verilator 5.035 devel rev v5.034-93-ged1e3ec4a

> What OS and distribution are you using?

WSL2 Ubuntu 24.04

[verilator_segfault.tar.gz](https://github.com/user-attachments/files/19532965/verilator_segfault.tar.gz)

wsnyder, 2025-03-31:
@kbieganski et al, could you take a look please, it's in the coroutine fork handling.

From `--runtime-debug` on non-WSL Ubuntu 22.04:

```
idx           0
idx           1
=================================================================
==1189387==ERROR: AddressSanitizer: heap-use-after-free on address 0x6130000036a8 at pc 0x5624fa92d3b0 bp 0x7ffe6dd30310 sp 0x7ffe6dd30300
READ of size 8 at 0x6130000036a8 thread T0
    #0 0x5624fa92d3af in Vuart_tb___024root___eval_initial__TOP__Vtiming__2 /verilator/verilator_segfault/obj_dir/Vuart_tb___024root__DepSet_h299ff880__0.cpp:53
    #1 0x5624fa887f32 in std::__n4861::coroutine_handle<void>::operator()() const /usr/include/c++/11/coroutine:124
    #2 0x5624fa887f32 in VlCoroutineHandle::resume() /verilator/include/verilated_timing.cpp:41
    #3 0x5624fa9263ed in Vuart_tb___024root___eval_initial__TOP__Vtiming__2____Vfork_1__0 /verilator/verilator_segfault/obj_dir/Vuart_tb___024root__DepSet_h299ff\
880__0.cpp:791
    #4 0x5624fa887f32 in std::__n4861::coroutine_handle<void>::operator()() const /usr/include/c++/11/coroutine:124
    #5 0x5624fa887f32 in VlCoroutineHandle::resume() /verilator/include/verilated_timing.cpp:41
    #6 0x5624fa88db1c in VlDelayScheduler::resume() /verilator/include/verilated_timing.cpp:65
    #7 0x5624fa8c962a in Vuart_tb___024root___timing_resume(Vuart_tb___024root*) /verilator/verilator_segfault/obj_dir/Vuart_tb___024root__DepSet_h299ff880__0.cpp:2068
    #8 0x5624fa8cf7d0 in Vuart_tb___024root___eval_phase__act(Vuart_tb___024root*) /verilator/verilator_segfault/obj_dir/Vuart_tb___024root__DepSet_h299ff880__0.cpp:2107
    #9 0x5624fa8cfc0b in Vuart_tb___024root___eval(Vuart_tb___024root*) /verilator/verilator_segfault/obj_dir/Vuart_tb___024root__DepSet_h299ff880__0.cpp:2166
    #10 0x5624fa92f2b2 in Vuart_tb::eval_step() /verilator/verilator_segfault/obj_dir/Vuart_tb.cpp:56
    #11 0x5624fa93e82b in Vuart_tb::eval() /verilator/verilator_segfault/obj_dir/Vuart_tb.h:55
    #12 0x5624fa93e82b in main /verilator/verilator_segfault/obj_dir/Vuart_tb__main.cpp:21
    #13 0x7f8ff3109d8f in __libc_start_call_main ../sysdeps/nptl/libc_start_call_main.h:58
    #14 0x7f8ff3109e3f in __libc_start_main_impl ../csu/libc-start.c:392
    #15 0x5624fa7d45d4 in _start (/verilator/verilator_segfault/obj_dir/Vuart_tb+0x41e5d4)

0x6130000036a8 is located 40 bytes inside of 344-byte region [0x613000003680,0x6130000037d8)
freed by thread T0 here:
    #0 0x7f8ff3d33d07 in operator delete(void*) ../../../../src/libsanitizer/asan/asan_new_delete.cpp:160
    #1 0x5624fa92ddce in Vuart_tb___024root___eval_initial__TOP__Vtiming__2 /verilator/verilator_segfault/obj_dir/Vuart_tb___024root__DepSet_h299ff880__0.cpp:94
    #2 0x5624fa890716 in VlCoroutineHandle::~VlCoroutineHandle() /verilator/include/verilated_timing.h:136

previously allocated by thread T0 here:
    #0 0x7f8ff3d331e7 in operator new(unsigned long) ../../../../src/libsanitizer/asan/asan_new_delete.cpp:99
    #1 0x5624fa92e3fd in Vuart_tb___024root___eval_initial__TOP__Vtiming__2(Vuart_tb___024root*) /verilator/verilator_segfault/obj_dir/Vuart_tb___024root__DepSet_h299ff880__0.cpp:94

SUMMARY: AddressSanitizer: heap-use-after-free /verilator/verilator_segfault/obj_dir/Vuart_tb___024root__DepSet_h299ff880__0.cpp:53 in Vuart_tb___024root___eval_initial__TOP__Vtiming__2
```

kbieganski, 2025-04-04:
I think the problem is we're overwriting the fork sync which is a module-level variable.
This should fix it:
```diff
diff --git i/src/V3Timing.cpp w/src/V3Timing.cpp
index c586365af..09092095b 100644
--- i/src/V3Timing.cpp
+++ w/src/V3Timing.cpp
@@ -723,8 +723,8 @@ class TimingControlVisitor final : public VNVisitor {
     void makeForkJoin(AstFork* const forkp) {
         // Create a fork sync var
         FileLine* const flp = forkp->fileline();
-        // If we're in a function, insert the sync var directly before the fork
-        AstNode* const insertBeforep = m_classp ? forkp : nullptr;
+        // Insert the sync var directly before the fork
+        AstNode* const insertBeforep = forkp;
         addCLocalScope(flp, insertBeforep);
         AstVarScope* forkVscp
             = createTemp(flp, forkp->name() + "__sync", getCreateForkSyncDTypep(), insertBeforep);
```

wsnyder, 2025-04-04:
@kbieganski much thanks!

closed 2025-04-04 by wsnyder

srcejon, 2025-04-04:
This fixes a couple of sims more complicated than the test case - thanks!