verilator/verilator #5912: Support 1800-2023 soft unions

rroohhh, 2025-04-03:
Running verilator from commit 14e7b1076e446226cabc5831d819813d1c828dfc against
```sv
typedef union soft packed {
    logic a;
    bit a;
} test_t;
```
produces the following error: 
```
%Error-UNSUPPORTED: /tmp/test.sv:1:15: Unsupported: 'union soft'
    1 | typedef union soft packed {
      |               ^~~~
                    ... For error description see https://verilator.org/warn/UNSUPPORTED?v=5.035
%Error: Exiting due to 1 error(s)
        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
```

wsnyder, 2025-04-03:
There's three things here:
1. Handle and test parsing `union soft` and `union soft packed`.  Verilator's algorithm I think "just works" if these keywords are ignored.
2. Add the IEEE-required error on unpacked hard when the sizes of union members don't match, and test it.

Might you consider making a pull to do these?

rroohhh, 2025-04-03:
Yeah, I'll take a jab at implementing this.

PR #5932 (2025-04-12): Support soft unions (#5912)
Fixes #5912
closed 2025-04-12 by wsnyder