verilator/verilator #5919: Handling tool quirks with define macros based on VHDL standard conditional analysis identifiers

jeras, 2025-04-05:
This is just a discussion.

The point is, the VHDL standard defines system variable names (would be define macros in SystemVerilog) for quirk workarounds. I am not saying they should be part of Verilator, it is just that since there is a standard there is no need to think of different names for SystemVerilog tools.

I use a lot of SystemVerilog and `X` propagation in my RTL/testbench code, so I often encounter various tool quirks. A common approach for handling them is to write workaround code for the quirky tool, and choosing between the default and the workaround with a macro defined for the specific tool. I often struggled finding a good name for the macro, so each project uses a different macro.

Yesterday I encountered a quirk in questa, any `X` bit in the operands to an addition propagates not only to the bits above the `X` operand bit, but to the entire result. I do not think the Vivado simulator does the same, but this would be 2 years ago, so I am not sure. ...

So I remembered that VHDL-2019 specifies "standard conditional analysis identifiers": `VHDL_VERSION`, `TOOL_TYPE`, `TOOL_VENDOR`, `TOOL_NAME`, `TOOL_EDITION`, `TOOL_VERSION`. I added them to my tool specific Makefiles. Now I do not have to invent a new macro name each time I wish to write a quirk workaround.

wsnyder, 2025-04-05:
Reference: https://www.intel.com/content/www/us/en/docs/programmable/683236/24-1/vhdl-2019-conditional-analysis-tool.html

While a good idea, I don't see how these can be used. Unfortunately IEEE, despite many suggestions to the contrary, decided not to support the "C" style "`ifdef TOOL_VERSION >= 1.234" which is what is really needed for this to be useful.   Without this many of the differences cannot be conditionalized (such as parse/elaboration differences).

So I guess I'm not understanding how you are using these with Verilator/Makefiles.  Maybe only in C?

jeras, 2025-04-05:
For me the important part is the `TOOL_NAME`. When I encounter a quirk, I usually waste some time googling whether the tool defines a macro I could use in the code to write a workaround for the given tool. Some tools actually have predefined macros, for example [Quartus Prime synthesis defines `ALTERA_RESERVED_QIS`](https://stackoverflow.com/questions/58996919/how-to-determine-that-synthesis-is-done-in-quartus). When I do not find an existing macro I waste time making up a name. The next time I will encounter this issue, I will just use the names defined for VHDL-2019.

This is my `questa` specific Makefile, I just pass the macros as CLI arguments, I will do the same for Verilator.
https://github.com/jeras/rp32/blob/master/sim/questa/Makefile.degu#L73-L94

Questa responds to the `-version` argument with a long string, which also depends on the command (`qrun`, `qverilog`, `vsim`) itself. I did not figure how to parse it yet (probably `sed`), but anyway, I just need `TOOL_NAME` in most cases.
```
$ qrun -version
Questa Intel Starter FPGA Edition-64 qrun 2024.1 Utility 2024.04 Apr 19 2024
Start time: 14:43:46 on Apr 05,2025
qrun -version 
```

Why I posted this as a Verilator issue? Because I am sure I searched the Verilator documentation for predefined macros many times before, and I might have already asked about it in a previous issue. Maybe not a great reason, but I was happy I now have a solution where I do not have to think about what name to give to the macro. I probably got overly enthusiastic over a small improvement in my development process. Reddit FPGA forum would probably be a better place to post this.

jeras, 2025-04-05:
Right now I am in the middle of updating a project after a 2 year hiatus. I started by running RISCOF on Questa. Vivado crashed, I will write an issue this week. So I only have an old example with Quartus Prime synthesis not supporting SV unions.

I used the `ALTERA_RESERVED_QIS` to define `LANGUAGE_UNSUPPORTED_UNION` (so I don't forget what the quirk was), and then used the later across the code.

https://github.com/jeras/rp32/blob/master/hdl/rtl/riscv/riscv_isa_i_pkg.sv#L19-L21

https://github.com/jeras/rp32/blob/master/hdl/rtl/riscv/riscv_isa_i_pkg.sv#L422-L506

I could now replace `ALTERA_RESERVED_QIS` with `TOOL_NAME == "quartus"` (not sure about the command name).

`TOOL_TYPE` (`"SIMULATION"`, `"SYNTHESIS"`, `"FORMAL"`) can be used to switch between RTL and fast simulation models.

Regarding `TOOL_EDITION`, it would probably be something defined by and IDE vendor using Verilator (or other open source EDA tools) as a component.

I am sure I will encounter more quirks as I port the code to Verilator, Vivado simulator and Xilinx/Altera/Lattice/Yosys? synthesis.

**Thanks for the documentation link.** I am not sure why, but I always spend too much time looking for this predefined macros (not just for Verilator). The VHDL-2019 approach seems to me like a good option to handle this the same way across various open source tools.

wsnyder, 2025-04-05:
> I could now replace ALTERA_RESERVED_QIS with TOOL_NAME == "quartus"

No, you can't in SystemVerilog, the SystemVerilog preprocessor doesn't do `==`, that's my point.

I realized that the term "predefined macro" didn't occur in the docs and fixed that.

Anyhow, I think we're agreed it's useful, but nothing more to directly do here (unless I missed something.)

Closing but feel free to follow up.

closed 2025-04-05 by wsnyder

jeras, 2025-04-05:
Sorry, you are right. I skimmed the SystemVerilog 2023 standard and remembered something about more powerful macros and erroneously thought I could do it. Still this time I will write a README so I don't have to search for this info again next time.

wsnyder, 2025-04-05:
Ok, also pulls to improve Verilator docs to help searchability or otherwise are always appreciated.