verilator/verilator #5920: verilator --coverage producing bad .cpp with code:  0U = 0U;

drewranck, 2025-04-06:
Thanks for taking the time to report this.

Can you attach an example that shows the issue?  (Must be openly licensed, completely self-contained and self-checking so can directly run what you provide. Ideally use test_regress format, see https://veripool.org/guide/latest/contributing.html?highlight=test_regress#reporting-bugs

> I will try, don't have a testcase I can share yet.

What 'verilator' command line do we use to run your example?

> long command, but essentially `verilator [--binary | --cc] --coverage`

What 'verilator --version' are you using?  Did you try it with the git master version?
> Verilator 5.033 devel rev v5.032-28-gbb871728c

What OS and distribution are you using?
> Ubuntu22.04

May we assist you in trying to fix this in Verilator yourself?
> Yes, but I first need to need a smaller distilled reproducible testcase first.

> Happens in my a function `V{{top}}..._eval_static__TOP`, where a bunch of values are assigned with `0U` in C++. A few it appears the left hand side didn't get resolved correctly, and so I get `0U = 0U;`  I suspect this is a struct, might be unpacked but not sure yet because is part of a big design. Also, it doesn't seem to affect all of my designs, just a few, and it's definitely in Testbench non-synthesizable code. SV-Interfaces are not involved.

Ideally a short term fix for me is - avoid this line `0U = 0U;` of .cpp entirely, even if it does affect coverage for this .sv line of code.

(Please avoid attaching screenshots that show text - you can convert images to text using e.g. https://ocr.space)

wsnyder, 2025-04-06:
Rather than `--coverage` please see which type of coverage is causing it: `--coverage-line`  `--coverage-toggle` `--coverage-expr` `--coverage-user`.  Likely the new expression coverage.

drewranck, 2025-04-07:
Thanks for the quick reply, and yeah it would probably help if I actually read the verilator --help. It appears to be `--coverage-line` causing this, works with the other coverage knobs added:

```
    --coverage-toggle \
    --coverage-underscore \
    --coverage-user \
    --coverage-max-width 512 \
```

drewranck, 2025-04-08:
Looks like this happens on a module instance with an input port connected to a constant, but I'll have to get this into a small enough testcase:

For example:

```
logic tie0, out1, out2;
assign tie0 = 1'b0;

somedut u_somedut (.in(1'b0), .out(out1)); // bad, input port mapped to a constant
somedut u_somedut2 (.in(tie0), .out(out2)); // good, input port mapped to tie0 
```

wsnyder, 2025-04-08:
See [Reporting Verilator Bugs](https://veripool.org/guide/latest/contributing.html?\
highlight=test_regress#reporting-bugs) and especially section on use of sv-bugpoint.  Should be straightfoward to have it search for the "0=0" error.

drewranck, 2025-04-08:
Thanks Wilson - I will try out sv-bugpoint, but I think I have a small example now.

Also, I think I was the one who broke this when I introduced support for default values on module input ports - https://github.com/verilator/verilator/issues/5358 - since my problem also goes away if I remove all default input values.

Works w/ `--coverage-line`:

```
module some_dut
  (
   input  in,
   output out
   );

  assign out = ~in;
endmodule : some_dut

module test_some_dut;

  logic out1;
  some_dut u_some_dut1 (.in(1'b1), .out(out1));

  initial begin
    #10ns;
    $finish;
  end
endmodule : test_some_dut
```

Fails with .cpp line `1U = 0U;` with `--coverage-line`:

```
module some_dut
  (
   input  in = 1'b0,
   output out
   );

  assign out = ~in;
endmodule : some_dut

module test_some_dut;

  logic out1;
  some_dut u_some_dut1 (.in(1'b1), .out(out1)); // fails if I set .in(1'b0) too.

  initial begin
    #10ns;
    $finish;
  end
endmodule : test_some_dut
```

wsnyder, 2025-04-09:
Well, I was expecting after your comment to punt to you to fix, but I see it's really failing in the module inliner; `no-inline` was a workaround.  But simple enough to resolve, fixed.

closed 2025-04-09 by wsnyder

drewranck, 2025-04-09:
Confirmed working for me on `Verilator 5.035 devel rev v5.034-154-g6bb57e463`