verilator/verilator #5927: Bad C++ (error: ‘vlSelf’ was not declared in this scope)

drewranck, 2025-04-09:
Thanks for taking the time to report this.

Can you attach an example that shows the issue?  (Must be openly licensed, completely self-contained and self-checking so can directly run what you provide. Ideally use test_regress format, see https://veripool.org/guide/latest/contributing.html?highlight=test_regress#reporting-bugs

> Working on it, sadly in another big codebase top level test.

What 'verilator' command line do we use to run your example?

> `verilator --binary --timing --assert --autoflush -j 2 -sv -Wno-CASEINCOMPLETE -Wno-REALCVT -Wno-SELRANGE -Wno-TIMESCALEMOD -Wno-UNSIGNED -Wno-WIDTH -CFLAGS -O0 --unroll-count 4 -top XX_YY_test -o sim.exe <defines, incdirs, files>`

What 'verilator --version' are you using?  Did you try it with the git master version?

> Fails on today's master: `Verilator 5.035 devel rev v5.034-154-g6bb57e463`
> Works on `Verilator 5.034 2025-02-24 rev v5.034`

What OS and distribution are you using?

> Ubuntu22.04

May we assist you in trying to fix this in Verilator yourself?

Sure. 

(Please avoid attaching screenshots that show text - you can convert images to text using e.g. https://ocr.space)

My actual (somewhat obfuscated) message looks like this:

```

VXX_YY_test___024root__DepSet_hc51b8287__0.cpp:172:9: error: ‘vlSelf’ was not declared in this scope
  172 |         vlSelf->__Vtask_XX_YY_test__DOT__uDUT__DOT__LongHierPath__unnamedblk1_1__DOT____Vrepeat0 = VL_RAND_RESET_I(32);
      |         ^~~~~~
VXX_YY_test___024root__DepSet_hc51b8287__0.cpp: In member function ‘VlCoroutine VXX_YY_test___024root::__VnoInFunc_XX_YY_test__DOT__UserCsrRead32_TOP(VXX_YY_test__Syms*, IData, IData&, IData)’:
VXX_YY_test___024root__DepSet_hc51b8287__0.cpp:1416:9: error: ‘vlSelf’ was not declared in this scope
 1416 |         vlSelf->__Vtask_XX_YY_test__DOT__uDUT__DOT__LongHierPath__unnamedblk1_4__DOT____Vrepeat3 = VL_RAND_RESET_I(32);
      |         ^~~~~~

```

I noticed in 5.034 (working version for me) most of these vlCoroutine .cpp functions have *vlSelf as an arg, and it's missing for some reason in 5.035. I'm not sure where to even start on this, it's a part of Verilator's src I have never explored. 

This could also be an obscure SV coding style issue on my end. Not all of these involve VL_RAND_RESET_I(32), and I didn't see a verilator knob for random reset values (I saw a runtime knob on the verialted exe, but we're not getting that far w/ bad .cpp).

drewranck, 2025-04-09:
Some more details -

It appears to be from a SV task automatic:

```
VlCoroutine VXX_YY_test___024root::__VnoInFunc_XX_YY_test__DOT__UserCsrWrite32_TOP(VXX_YY_test__Syms* __restrict vlSymsp, IData/*31:0*/ address, IData/*31:0*/ data, IData/*31:0*/ number) {
```

from SV code: 

```
  task automatic UserCsrWrite32(input [31:0] address, input [31:0] data, input int number=-1);
```

Looking at 5.034 (working), all of these similar looking statements use `this->` instead of 5.035 (problems) which have a mixed bag of `vlSelf->` and `this->`. I'm a complete newb here, but looking at the names of the .cpp VlCoroutine methods - the ones that have a `VXX_YY_test___024root* vlSelf` arg are for the SV `initial` blocks, and the tasks do not have it.

So, sure it's just as easy as making vlSelf-> into this-> :) piece of cake. Grepping around in verilator src and I'm not seeing any differences at all between 5.034 and master.

wsnyder, 2025-04-09:
See here, maybe what generates the suspect function is missing isLoose?  Look at the .tree files to see.

https://github.com/verilator/verilator/blob/6bb57e463035a94fafaefe652e747303f9c548e0/src/V3EmitCBase.cpp#L116

drewranck, 2025-04-09:
here's the start of the tree.json for this SV task:

```
       {
          "type": "TASK",
          "name": "UserCsrWrite32",
          "addr": "(UPD)",
          "loc": "eo,494:18,494:32",
          "editNum": 208947,
          "method": false,
          "dpiExport": false,
          "dpiImport": false,
          "dpiOpenChild": false,
          "dpiOpenParent": false,
          "isExternDef": false,
          "isExternProto": false,
          "prototype": false,
          "recursive": false,
          "taskPublic": false,
          "cname": "UserCsrWrite32",
          "fvarp": [],
          "classOrPackagep": [],
          "stmtsp": [
```

Looking in V3Task.cpp, is seems I won't get isLoose set:

```
        if (nodep->dpiImport() || nodep->dpiExport()) {
            cfuncp->isStatic(true);
            cfuncp->isLoose(true);
        } else {
            cfuncp->isStatic(false);
        }
````

As a hacker attempt, I manually changed the generated .cpp files that had `vlSelf->` when function arg vlSelf wasn't present I replaced with `this->`, and ran `make` using the .mk file. It compiles and sim.exe ran cleanly. So I don't know that I want isLoose(true) to happen.

I'm not even using these tasks as something that's accessible via DPI. These are simply internal SV tasks called by SV. Is that what's happening here?

Since I don't know what I'm doing, I also tried changing V3Task.cpp to have isLoose(true) (for all tasks, why not?), which did add the function arg for vlSelf, but this caused more spectacular .cpp compile fails, I suspect whoever called this C++ function didn't know this arg was needed. What does isLoose() mean? Not attached to a package, class, or something else?

drewranck, 2025-04-10:
I'm having a really hard time trying to fix this one. 

I tried passing some flag through to AstCReset - so when it was visited (V3EmitCFunc.h) I could tell `emitVarReset` to prefer using `this` over `vlSelf`, but that didn't go over well.

I'm almost inclined to, if the generated .cpp function args don't include `* vlSelf`, then declare `vlSelf` in function body and assign to `this`. This also appeared to work, added to each function manually and re-ran make, but I can't quite figure out how to do this easily in V3EmitCFunc or V3EmitCBase:

`VXX_YY_test___024root* vlSelf = this;`

wsnyder, 2025-04-10:
Can you please try sv-bugpoint to make a small test you can share?

drewranck, 2025-04-10:
Manually whittled down to a failing testcase:

beep_boop_test.sv
```

module beep_boop_test;

  logic  clock;

  initial begin
    clock = '0;
    /* verilator lint_off INFINITELOOP */
    forever begin
      clock = #5ns ~clock;
    end
    /* verilator lint_on INFINITELOOP */
  end

  task static beeps_has_fork_task (input [63:0] address);
    @(posedge clock);
    // Workaround0: Comment this 'fork' out to pass the compile.
    //   You can leave the repeat - clock stmt.
    fork
      begin : boops_and_beeps
        repeat($urandom_range(9)) @(posedge clock);
      end
    join
  endtask : beeps_has_fork_task

  // I've intentionally created a recursive task chain (that should be unreachable anyway):
  // call_the_booper_task()
  //   --> (unreachable) --> calls local_booper_sub_task()
  //       --> calls call_the_booper_task()
  //             --> ...
  //   --> (reachable) --> calls beeps_has_fork_task() done.

  // Changing tasks from automatic to static has no effect.

  task static call_the_booper_task(input [31:0] address);
    if (1) begin
      // Workaround1: Comment this out to pass the compile.
      beeps_has_fork_task(address);
    end
    else begin
      // Workaround2: Comment this out to pass the compile
      // Should be unreachable anyway.
      local_booper_sub_task(.address(address));
    end
  endtask : call_the_booper_task

  task static local_booper_sub_task(input [31:0] address);
    logic [63:0] req;
    logic [39:0] resp;
    req = '0;
    call_the_booper_task(.address(32'h0000_1234));
    resp = '0;
  endtask : local_booper_sub_task

  initial begin : main
    $info("start");
    #100ns;
    $finish;
  end

endmodule : beep_boop_test
```

verilator command:
```
verilator --binary --timing -sv \
          -Wno-WIDTH \
          -top beep_boop_test \
          -o sim.exe \
          beep_boop_test.sv
```

drewranck, 2025-04-10:
I should also say, it doesn't matter if the tasks and `static` or `automatic`.

Fail seen:
```
In file included from Vbeep_boop_test__ALL.cpp:5:
Vbeep_boop_test___024root__DepSet_h558e95e6__0.cpp: In member function ‘VlCoroutine Vbeep_boop_test___024root::__VnoInFunc_beep_boop_test__DOT__call_the_booper_task_TOP(Vbeep_boop_test__Syms*, IData)’:
Vbeep_boop_test___024root__DepSet_h558e95e6__0.cpp:11:5: error: ‘vlSelf’ was not declared in this scope
   11 |     vlSelf->__Vtask_beep_boop_test__DOT__beeps_has_fork_task__0__beep_boop_test__DOT__boops_and_beeps__DOT__unnamedblk1_1__DOT____Vrepeat0 = VL_RAND_RESET_I(32);
      |     ^~~~~~
make: *** [/usr/local/share/verilator/include/verilated.mk:273: Vbeep_boop_test__ALL.o] Error 1
```

closed 2025-04-26 by wsnyder