verilator/verilator #5933: Support Force/Release via VPI

ktbarrett, 2025-04-11:
There's functionality for making a signal forceable (pragma `/* verilator forceable */` or `forceable -module {} -var {}` on the commandline). This creates special signals for implementing Force/Release logic. However, the VPI directly supports using `vpiForce` and `vpiRelease` when setting values on signals and this isn't supported right now.

I propose that VPI writes to signals marked as forceable use these special signals instead of the normal signal reference to ensure that Force/Release semantics to exist.

wsnyder, 2025-04-11:
I agree this makes sense.  A pull would be appreciated, as it's unlikely to get addressed soon otherwise.

ktbarrett, 2025-04-15:
For those others concerned, it's on my radar, but lower priority. Help is appreciated.