verilator/verilator #5941: Modport interface field access via hierarchy fails

hankhsu1996, 2025-04-17:
This design causes Verilator to error when accessing a signal through a modport-connected interface via hierarchy. Other tools like slang and VCS accept it without issues.

### Testcase

```systemverilog
interface bus_if(input logic clk);
  logic [7:0] data;
  modport slave(output data, input clk);
endinterface

module slave(bus_if.slave bus);
  always_ff @(posedge bus.clk)
    bus.data <= 8'h5A;
endmodule

module top;
  logic clk = 0;
  always #5 clk = ~clk;

  bus_if bus(clk);
  slave slave_inst(bus);

  initial begin
    #20;
    $display("bus.data = %h", bus.data);                       // OK
    $display("slave_inst.bus.data = %h", slave_inst.bus.data); // Fails
    $finish;
  end
endmodule
```

### Command

```
verilator --binary -j 0 --top-module top test.sv
```

### Output

```
%Error: test.sv:21:57: Can't find definition of 'bus' in dotted signal: 'slave_inst.bus.data'
   21 |     $display("slave_inst.bus.data = %h", slave_inst.bus.data);
      |                                                         ^~~~
        ... Known scopes under 'data': <no instances found>
%Error: Verilator internal fault, sorry. Suggest trying --debug --gdbbt
```

### Versions tested

- Verilator 5.028  
- Verilator 5.034 (latest)  
Both show the same error.

### OS

Ubuntu 22.04.4 LTS (WSL2)

### Notes

Not 100% sure if this is technically valid per the LRM, but other tools like slang and VCS compile and simulate it without issue. At the very least, Verilator should not crash internally when encountering this pattern.

wsnyder, 2025-04-17:
I believe this is legal. Looks like V3LinkDot hasn't seen this before.  Perhaps you'd attempt a pull request to fix it?

hankhsu1996, 2025-04-18:
Ok, I'll try