verilator/verilator #5943: UNOPTFLAT reports should mention isolate_assignments

aquohn, 2025-04-18:
Apparent combinational loops can arise from variables that need to be split, but also from overly large always blocks with many assignments. The former is solved with `split_var` and the latter with `isolate_assignments`, but the warnings that come up with UNOPTFLAT only mention the former.

I've written a simple patch for the warning to mention `isolate_assignments` (#5942), but might it be preferrable to incorporate some logic for detecting loops which might be broken by it?

wsnyder, 2025-04-18:
Pull merged.  I'm not sure what you are suggesting?  It already shows loops, that is what UNOPTFLAT means.

aquohn, 2025-04-18:
What I had in mind was an analysis like the one done for `split_var`, perhaps to list high-fanout `always @(*)` blocks. But I'm not familiar with Verilator's internal data model, so I don't know how feasible that would be.

wsnyder, 2025-04-18:
I understand how that might be useful, but I'm going to close I don't think we're likely to get to that, especially as we now do split_var mostly by default.

closed 2025-04-18 by wsnyder