verilator/verilator #5945: Error on normal comment starting with Verilator

hankhsu1996, 2025-04-18:
I use multiple simulators in my project, including Verilator. It's common to leave comments about simulator-specific behavior. For example:

```systemverilog
// Verilator does not support class in generate block yet
`ifndef VERILATOR
  // ...
`endif
```

In this case, Verilator reports an error:

```
Unknown Verilator comment: '/*Verilator does not support class in generate block yet*/'
```

This comment is not intended as a directive, just a note for developers. But because it starts with "Verilator", it gets picked up by Verilator's directive parser and results in an error.

I've run into this several times and had to rephrase the comment to work around it. That feels a bit off, since a regular comment shouldn't have any effect on how the tool behaves.

I'd like to suggest:

1. Treat unknown Verilator comments as warnings instead of errors, so users can choose to ignore them
2. Only treat comments as directives if they start with `verilator <directive>` and the directive is recognized

Would love to hear what you think.

wsnyder, 2025-04-18:
>Only treat comments as directives if they start with verilator <directive> and the directive is recognized

How do we know it's not a mistyped directive?  That's the intent of this.

>Treat unknown Verilator comments as warnings instead of errors, so users can choose to ignore them

That's reasonable, might you make a pull to do this?

keesj, 2025-04-23:
I also just unexpectedly hit the same issue while documenting some code. I ended up rewording as I also would not like to see those warning anyway. That said I am getting used to the -Wno-SOMETHING hence this is where I would prefer to to see a similar option for this

closed 2025-04-29 by wsnyder