verilator/verilator #5947: Windows: sanitize paths from backslashes to '/' or use '\\'

Kreijstal, 2025-04-18:
This is with verilator master version. 

I get the following warning

```
Vsystem_tb___024root__DepSet_h3fe66a9b__0.cpp:97688:66: warning: unknown escape sequence: '\j'
97688 |                                                                  "C:/Users/kreij/scoop/apps/msys2/2022-01-28/home/kreij/ok/risc-v-lab/src/fv\jtag_master.sv",
      |    
```

I know this is not a terribly bad bug, but I get a whole screen of these warnings it would be nice if in windows paths could get sanitized to '\', I am compiling verilator using the msys2 toolchain on windows 10.

wsnyder, 2025-04-18:
Seems a good idea. Please consider making a pull to clean this up.  Most here don't use Windows so it's unlikely to get attention otherwise.

closed 2025-04-29 by wsnyder

Kreijstal, 2025-04-30:
wow thank you, I still havne't yet learned verilator codebase so sorry for not doing a pr.