verilator/verilator #5949: nested systemverilog interface

ganghuang, 2025-04-19:
I am using cocotb with verilator to simulate systemverilog code. Verilator is one of the most systemverilog supported open source simulator. But I got error when try to simulate nested systemverilog interface. 
I can not tell this is a bug, or feature request, so I just ask in the Q&A here. 

A simplified example below:
The .sv: There are 4 different ways I want to use the interface, only directly instantiate it at top level (test1) can be compiled and run successfully. If I want to access the sub-interface in the submodule, it report error. 
```
interface ift5(input clk);
    reg [31:0] cnt=0;
endinterface : ift1

interface ift4(input clk);
    ift5 tif5(.clk(clk));
endinterface : ift4

interface ift3(input clk);
    ift4 tif4(.clk(clk));
endinterface : ift3

interface ift2(input clk);
    ift3 tif3(.clk(clk));
endinterface : ift3

interface ift1(input clk);
    ift2 tif2(.clk(clk));
endinterface : ift1

module m1(ift1 tif1,input clk);
m3 m3(.tif3(tif1.tif2.tif3),.clk(clk));
endmodule:m1

module m2(ift2 tif2,input clk);
always @(posedge tif2.clk) begin
        tif2.tif3.tif4.tif5.cnt<=tif2.tif3.tif4.tif5.cnt+1'b1;
end
endmodule:m2

module m3(ift3 tif3,input clk);
always @(posedge tif3.clk) begin
        tif3.tif4.tif5.cnt<=tif3.tif4.tif5.cnt+1'b1;
end
endmodule:m3

module plsvsim(input clk
);
ift1 tif1(.clk(clk));
ift3 tif3(.clk(clk));
// test 1 can pass
m3 m3(.tif3(tif3),.clk(clk));
// test 2 fail
//m1 m1(.tif1(tif1),.clk(clk));
// test 3 fail
//m3 m3(.tif3(tif1.tif2.tif3),.clk(clk));
// test 4 fail
//m2 m2(.tif2(tif1.tif2),.clk(clk));
endmodule

```
And the .py:
```
import cocotb
from cocotb.triggers import Timer, RisingEdge
from cocotb.clock import Clock
import numpy as np
import json

@cocotb.test()
async def plsvsim(dut):
    clock = Clock(dut.clk,10,units="ns")
    cocotb.start_soon(clock.start())
    for i in range(10):
        await RisingEdge(dut.clk)
```
When I run make for the cocotb, it call verilator
```
rm -f results.xml
"make" -f Makefile results.xml
make[1]: Entering directory '/home/user/pstest/sim'
/usr/local/bin/verilator -cc --exe -Mdir sim_build -DCOCOTB_SIM=1 --top-module plsvsim -Wno-fatal --timescale 1ns/1ps --vpi --public-flat-rw --prefix Vtop -o Vtop -LDFLAGS "-Wl,-rpath,/home/user/venv/lib/python3.12/site-packages/cocotb/libs -L/home/user/venv/lib/python3.12/site-packages/cocotb/libs -lcocotbvpi_verilator" -sv -error-limit 5 --hierarchical -j 8 --trace --trace-structs --trace-fst -Wno-PINMISSING -Wno-REALCVT plsvsim.sv /home/user/venv/lib/python3.12/site-packages/cocotb/share/lib/verilator/verilator.cpp
make -C sim_build  -f Vtop.mk
```
if I uncomment the lines for test2, test3, test4 in the sv file, it will complain:
```
rm -f results.xml
"make" -f Makefile results.xml
make[1]: Entering directory '/home/user/pstest/sim'
/usr/local/bin/verilator -cc --exe -Mdir sim_build -DCOCOTB_SIM=1 --top-module plsvsim -Wno-fatal --timescale 1ns/1ps --vpi --public-flat-rw --prefix Vtop -o Vtop -LDFLAGS "-Wl,-rpath,/home/user/venv/lib/python3.12/site-packages/cocotb/libs -L/home/user/venv/lib/python3.12/site-packages/cocotb/libs -lcocotbvpi_verilator" -sv -error-limit 5 --hierarchical -j 8 --trace --trace-structs --trace-fst -Wno-PINMISSING -Wno-REALCVT plsvsim.sv /home/user/venv/lib/python3.12/site-packages/cocotb/share/lib/verilator/verilator.cpp
%Error: Internal Error: plsvsim.sv:31:16: ../V3Scope.cpp:80: Can't locate varref scope
   31 | module m3(ift3 tif3,input clk);
      |                ^~~~
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
make[1]: *** [/home/user/venv/lib/python3.12/site-packages/cocotb/share/makefiles/simulators/Makefile.verilator:64: sim_build/Vtop.mk] Error 1 
make[1]: Leaving directory '/home/user/pstest/sim'
make: *** [/home/user/venv/lib/python3.12/site-packages/cocotb/share/makefiles/Makefile.inc:40: sim] Error 2 
```

closed 2025-04-19 by wsnyder

ganghuang, 2025-04-19:
Thank you for the quick response and point to a solved thread. 
But I do not get how was the problem fixed.
Compare the two zip file in the thread linked, 
```
 diff t_interface_nested_to_mod.v ../1/t_interface_nested_to_mod.v 
18c18
<    logic sig2;
---
>    logic sig2 = 1'b1;
20a21,26
>    bit   pass;
> 
>    initial begin
>       pass                 = 1'b1;
>       top_if1.if1_if2.s2   = 1'b0;
>    end
28,40c34,35
<       if (cntr == 10) begin
<          top_if1.if1_if2.s2 <= 1'b0;
<          sig2 <= 1'b1;
<       end else begin
<          top_if1.if1_if2.s2 <= !top_if1.if1_if2.s2;
<          sig2 <= !sig2;
<          if (sig1 != sig2) begin
<             $finish();
<          end else if (cntr == 0) begin
<             $write("*-* All Finished *-*\n");
<             $finish;
<          end
<       end
---
>       top_if1.if1_if2.s2 <= !top_if1.if1_if2.s2;
>       sig2 <= !sig2;
41a37,43
> 
>       if (sig1 != sig2) begin
>          $finish();
>       end else if (cntr == 0) begin
>          $write("*-* All Finished *-*\n");
>          $finish;
>       end
```
and 
```
diff t_interface_nested_gen_to_mod.v ../1/t_interface_nested_gen_to_mod.v 
53a54
>    initial  interface1Signal1 = 1'b0;
57a59
>    initial  interface2Signal1 = 1'b0;
```
That kind of hint the initialization is the difference. 
I add initialization in the module m1, it still complain the same:
```
interface ift5(input clk);
    logic [31:0] cnt;
    initial cnt=0;
endinterface : ift5

interface ift4(input clk);
    ift5 tif5(.clk(clk));
endinterface : ift4

interface ift3(input clk);
    ift4 tif4(.clk(clk));
endinterface : ift3

interface ift2(input clk);
    ift3 tif3(.clk(clk));
endinterface : ift2

interface ift1(input clk);
    ift2 tif2(.clk(clk));
endinterface : ift1

module m1(ift1 tif1,input clk);
initial begin
    tif1.tif2.tif3.tif4.tif5.cnt =0;
end
m3 m3(.tif3(tif1.tif2.tif3),.clk(clk));
endmodule:m1

module m2(ift2 tif2,input clk);
always @(posedge tif2.clk) begin
        tif2.tif3.tif4.tif5.cnt<=tif2.tif3.tif4.tif5.cnt+1'b1;
end
endmodule:m2

module m3(ift3 tif3,input clk);
always @(posedge tif3.clk) begin
        tif3.tif4.tif5.cnt<=tif3.tif4.tif5.cnt+1'b1;
end
endmodule:m3

module plsvsim(input clk
,output led
);
ift1 tif1(.clk(clk));
ift3 tif3(.clk(clk));
m3 m3(.tif3(tif3),.clk(clk));
m1 m1(.tif1(tif1),.clk(clk));
//m3 m3(.tif3(tif1.tif2.tif3),.clk(clk));
//m2 m2(.tif2(tif1.tif2),.clk(clk));
assign led=tif1.tif2.tif3.tif4.tif5.cnt[31];
endmodule
```

ganghuang, 2025-04-19:
@wsnyder 
Or you  mean the comment you given in the other thread from 2020 still valid:

"
Looked at this some, basically nested interfaces never worked and require some internal changes (from AstVarRef to AstVarXRef) and probably other stuff. Will give it some more time shortly.
"

That means the nested interface is still not supported in verilator?
Thanks

wsnyder, 2025-04-19:
Correct, the bug is not fixed, if it was the issue would be closed not open.