verilator/verilator #5951: Modport related issue

WajahatRiaz, 2025-04-21:
I wrote a simple driver task in my testbench. Which is basically assigning a value to a modport member. 
```
`define DRIV_IF vif.DRIVER
    task reset();
        `DRIV_IF.CHI_TXLINKACTIVEACK = 0;
        `DRIV_IF.CHI_RXLINKACTIVEREQ = 0;
        `DRIV_IF.CHI_TXRSPLCRDV = 0;
        `DRIV_IF.CHI_TXDATLCRDV = 0;
        `DRIV_IF.CHI_RXDATFLITPEND = 0;
        `DRIV_IF.CHI_RXDATFLITV = 0;
        `DRIV_IF.CHI_RXDATFLIT = 0;
        `DRIV_IF.CHI_RXREQFLITPEND = 0;
        `DRIV_IF.CHI_RXREQFLITV = 0;
        `DRIV_IF.CHI_RXREQFLIT = 0;
        `DRIV_IF.CHI_RXSACTIVE = 1;
    	
        wait(vif.reset);  // waiting for the reset to be asserted so that we can send stimulus to DUT

        $strobe("\nRESET APPLIED!");
        
        repeat (2) @(negedge vif.clk);  /* The earliest point after reset that the component is
                                               permitted to begin driving these signals HIGH is at a
                                               rising CLK edge after RESETn is HIGH */
    endtask
```

The modport definition: 
```
   modport DRIVER(
        input clk,
            reset,
            CHI_TXSACTIVE,
            CHI_TXLINKACTIVEREQ,
            CHI_RXLINKACTIVEACK,
            CHI_TXRSPFLITPEND,
            CHI_TXRSPFLITV,
            CHI_TXRSPFLIT,
            CHI_TXDATFLITPEND,
            CHI_TXDATFLITV,
            CHI_TXDATFLIT,
            CHI_RXDATLCRDV,
            CHI_RXREQLCRDV,

        output CHI_RXSACTIVE,
            CHI_TXLINKACTIVEACK,
            CHI_RXLINKACTIVEREQ, // FOR RX
            CHI_TXRSPLCRDV,
            CHI_TXDATLCRDV,
            CHI_RXDATFLITPEND,
            CHI_RXDATFLITV,
            CHI_RXDATFLIT,
            CHI_RXREQFLITPEND,
            CHI_RXREQFLITV,
            CHI_RXREQFLIT
    );
```
Verilator command:

```
verilator --binary -j 8 --skip-identical \
	--timing -Wno-WIDTHEXPAND -Wno-WIDTHTRUNC\
	-f "./sle-chi/flist.f" \
	"./sle-chi/src/chi_hn_top.sv" "./sle-chi/sim/top.sv" --top top -assert --x-assign 1 --x-initial unique \
	--autoflush --trace --coverage-line --coverage-user \
	--timescale-override 10ns/10ns
```

My solver verilator: z3
My verilator verison: Verilator 5.035 devel rev v5.034-27-g258becd17. It's a git master verison.
My OS and distrbution details: Ubuntu 24.04.2 LTS

LRM: IEEE Standard for SystemVerilog—Unified Hardware Design, Specification, and Verification Language

wsnyder, 2025-04-21:
Can you please provide a complete example that passes on other simulators?

WajahatRiaz, 2025-04-21:
@wsnyder that won't be possible.

WajahatRiaz, 2025-04-21:
@wsnyder Attaching screenshot here: 

[image]

WajahatRiaz, 2025-04-21:
Having knowledge about this error would help: 
../V3Width.cpp:2940: MemberSel of non-variable

WajahatRiaz, 2025-04-21:
@wsnyder just updated the issue.

wsnyder, 2025-04-21:
Perhaps #5941 though error is different.  It's unlikely though to get fixed without a complete test case, you can see 

For help in making a small test case see [Reporting Verilator Bugs](https://veripool.org/guide/latest/contributing.html?highlight=test_regress#reporting-bugs) section on use of sv-bugpoint.

Though to be honest even with a test case I'm not sure how soon I'll be able to fix it.... getting a pull to fix it would be most welcome.

wsnyder, 2025-05-18:
If you get an example you can share please reopen.

closed 2025-05-18 by wsnyder