verilator/verilator #5955: RTLMeter: OpenTitan Bug at cppbuild Stage

buhraian, 2025-04-22:
Hello @gezalore,

I was just glancing through the PR for adding in the RTLMeter Github workflow (cool work btw, I used to be one of Jon's undergrad researchers so it's awesome to see OpenPiton in there!) and saw that it was potentially my commit that broke the cppbuild stage for OpenTitan, so I wanted to investigate further to make sure that I didn't mess anything up. I tested multiple versions of Verilator pre commit (6bb57e4), immediately before commit (0359bf3), and post commit (50d7f2a), and found that I ran into the same 
```error: no matching function for call to ‘VL_UNPACK_WW(int, int, VlWide<3>&, VlWide<3>&)’```
specifically for the ``revcol`` function within ``prim_lfsr.sv`` with all these versions of Verilator. I believe the issue is actually localized to some interaction between ``CvtPackedToArray`` and the ``{<<{}}`` streaming operator, rather than what I introduced with ``CvtUnpackedtoQueue``. 

I attached a minimal (not self-checking) version of ``prim_lfsr.sv`` that demonstrates this behavior (when Verilated with the ``--trace`` flag so it doesn't get optimized away). Interestingly, when setting ``LfsrDw`` to something less than ``64``, Verilator will error with ``Verilator internal fault, sorry.`` and is not even able to reach the cppbuild stage. Unfortunately, I currently don't have a lot of time to go debug/fix this, but I wanted to open up this issue now for visibility/updates/preliminary investigation purposes until someone has the time to dive deeper. 

[prim_lfsr.sv.txt](https://github.com/user-attachments/files/19856497/prim_lfsr.sv.txt)

wsnyder, 2025-04-22:
I suspect it shouldn't be using CvtPackedToArrary in this case as it's a packed vs unpacked array, but need time to dig deeper, if you'd like to dig in it would be appreciated.

gezalore, 2025-04-23:
Thanks for looking into this. The OpenTitan case in RTLMeter used to work with Verilator around mid March, hence my [apparently incorrect] guess at the breaking commit. It should be fairly easy to bisect if anyone fancies, otherwise I will get to it at some point when I have a bit more time.

buhraian, 2025-04-29:
Just wanted to give a quick update, I narrowed the issue to commit 3694d15. I was playing around with changes to ``void visit(AstNodeAssign* nodep)`` in ``V3Width.cpp`` to prevent a new ``AstCvtPackedToArray`` from being emitted when the stream's ``lhsp``  is of ``PackArrayDType`` as well, and while this passed the ``t_stream_unpack`` test as well as my above test case, it breaks when the ``lhsp`` of the stream is a basic data type that has the same width. For example,

```
module test #(
    parameter WIDTH = 256
)(
    input [WIDTH-1:0] in,
    output [WIDTH-1:0] out
);
    logic [7:0][(WIDTH/8) - 1:0] key_init;

    assign key_init = {<<8{in}};
    assign out = key_init;
endmodule
```
will fail since ``in`` is of type ``logic [255:0]`` which means an ``AstCvtPackedToArray`` will still be created (improperly?).

I don't really have time this week to dig further, but I did want to give an update in case someone wanted to pick this back up.

gezalore, 2025-05-04:
> Just wanted to give a quick update, I narrowed the issue to commit [3694d15](https://github.com/verilator/verilator/commit/3694d153aeee66fbd30628702f9ae014f0a63a85). I was playing around with [...]

This test case works fine with 3694d15 for me, and the segfault seems like a different issue to what is causing OpenTitan to fail in RTLMeter. I am investigating OpenTitan, but won't look into this one for now.

gezalore, 2025-05-04:
You were on the right track after all, your minial example just shows a different failure mode for the same issue. Breaking patch was 6bb57e463035a94fafaefe652e747303f9c548e0.

buhraian, 2025-05-05:
The 6bb57e4 commit was what I meant to include in the update, not sure how/why I put another commit there 😅. Sorry about the confusion there, glad it was an easy fix. I appreciate the work put into RTLMeter as well, it has been very helpful for some other work I've been doing too, cheers 👍.

gezalore, 2025-05-05:
No worries.

> I appreciate the work put into RTLMeter as well, it has been very helpful for some other work I've been doing too

I would be interested to know what you are up to with RTLMeter. I was hoping it could be useful for others, not just for Verilator, so I would be keen to learn of improvement opportunities. You can find me in private if you prefer.

buhraian, 2025-05-05:
@gezalore What would be a good place/email to reach you at?

gezalore, 2025-05-05:
The email address you can find in all my git commits using `git log` should be fine

PR #5976 (2025-05-04): Fix streaming to/from packed arrays (#5955)
bug from 6bb57e463035a94fafaefe652e747303f9c548e0

Fixes RTLMeter OpenTitan. Fixes #5955.