verilator/verilator #5961: Result missmatched with vcs

jcteng, 2025-04-27:
What 'verilator' command line do we use to run your example?
``` verilator tb_zcounter.sv zcounter.sv  --binary --trace```

What 'verilator --version' are you using?  Did you try it with the git master version?
5.034 compiled with clang-18 

What OS and distribution are you using?
ubuntu20.04

May we assist you in trying to fix this in Verilator yourself?
Sure , but we need clarify reason first. 

After executing the simple bench below, the waveform data obtained shows inconsistencies between the results calculated by Verilator and VCS. According to common understanding, both Verilator and VCS comply with the IEEE Verilog standard. However, in actual simulations, they produce different waveform outputs, which is quite puzzling. What is the reason for this discrepancy? Or, how can such differences be avoided?

[image]

```verilog
module zcounter(
input clk,
input rstn,
input [7:0] d
);

reg [7:0]cnt;
initial begin
// #1 $display("reseted");
end

always @(posedge clk or rstn)
begin
    if(!rstn)
        cnt <= 0;
    else
        cnt <= cnt + 1;
end

logic [7:0] left_idx_cnt;
logic [7:0] left_idx_cnt_d;

always @(posedge clk or rstn) begin
    if(!rstn)
        left_idx_cnt_d <= 'h40;
    else
        left_idx_cnt_d<= left_idx_cnt;
end
logic [7:0] idx_mask_head;
logic [7:0] idx_mask_cnt;

wire [7:0] start_to_boundary_cnt= cnt-idx_mask_head;

always @* begin
    if(cnt<=6) begin
        idx_mask_head =8;
        left_idx_cnt=10;
        idx_mask_cnt=6;
    end else begin
        idx_mask_head = 0;
        left_idx_cnt = left_idx_cnt_d - idx_mask_cnt;
        idx_mask_cnt =start_to_boundary_cnt-1;
    end
end

endmodule
```

```verilog
`timescale 1ps/1ps

module tb_zcounter;

  // Inputs to DUT
  reg clk;
  reg rstn;
  reg [7:0] d;

  // Instantiate the Device Under Test (DUT)
  zcounter dut (
      .clk(clk),
      .rstn(rstn),
      .d(d)
  );

  // Clock generation (e.g., 100MHz = 10ns = 10000ps period -> 5000ps half-period)
  // Using 10ps half-period from the example image for consistency
  initial begin
    clk = 1;
    forever #10 clk = ~clk; // 20ps clock period
  end

  // Stimulus generation
  initial begin
    // Initialize signals
    rstn = 0; // Assert reset
    d = 0;

    // Deassert reset after some time
    #30 rstn = 1;

    // Apply input value after reset is deasserted
    #60 d = 3; // Apply d=3 at time 30ps + 60ps = 90ps

  end

  // Simulation control
  initial begin
    #1000; // Run simulation for 1000ps
    $finish;
  end

  // Waveform dumping
  initial begin
    $dumpfile("zcounter_wave.vcd");
    $dumpvars(0, tb_zcounter); // Dump all signals in this module and below
  end

endmodule
```

FanShupei, 2025-04-27:
```
always @(posedge clk or rstn)
begin
    if(!rstn)
        cnt <= 0;
    else
        cnt <= cnt + 1;
end
```

The coding style is pretty strange.
1. for sync reset, please use `always @(posedge clk)`
2. for async reset, please use `always @(posedge clk or negedge clk)`

jcteng, 2025-04-27:
@FanShupei updated 

no changes, gap still exists between VCS and verilator
``` verilog
module zcounter(
input clk,
input rstn,
input [7:0] d
);

reg [7:0]cnt;
initial begin
// #1 $display("reseted");
end

always @(posedge clk )
begin
    if(!rstn)
        cnt <= 0;
    else
        cnt <= cnt + 1;
end

logic [7:0] left_idx_cnt;
logic [7:0] left_idx_cnt_d;

always @(posedge clk ) begin
    if(!rstn)
        left_idx_cnt_d <= 'h40;
    else
        left_idx_cnt_d<= left_idx_cnt;
end
logic [7:0] idx_mask_head;
logic [7:0] idx_mask_cnt;

wire [7:0] start_to_boundary_cnt= cnt-idx_mask_head;

always @* begin
    if(cnt<=6) begin
        idx_mask_head =8;
        left_idx_cnt=10;
        idx_mask_cnt=6;
    end else begin
        idx_mask_head = 0;
        left_idx_cnt = left_idx_cnt_d - idx_mask_cnt;
        idx_mask_cnt =start_to_boundary_cnt-1;
    end
end

endmodule
```

[image]

FanShupei, 2025-04-27:
```
always @* begin
    if(cnt<=6) begin
        idx_mask_head =8;
        left_idx_cnt=10;
        idx_mask_cnt=6;
    end else begin
        idx_mask_head = 0;
        left_idx_cnt = left_idx_cnt_d - idx_mask_cnt; // swap with the next line
        idx_mask_cnt =start_to_boundary_cnt-1;
    end
end
```

May you try this change? The code has a synthesis/simulation mismatch. If the code indent to model a combinational logic, since `left_idx_cnt` has dependency on `idx_mask_cnt`, the assign of `idx_mask_cnt` must come earlier.

jcteng, 2025-04-27:
@FanShupei 
In fact, the code snippet mentioned earlier comes from fragments of a complex project. In real projects, when using Verilator for simulation, we hope it adheres to the IEEE System Verilog standard for simulation. In this snippet, VCS and Verilator exhibited different behaviors, suggesting that one of the simulators may not be complying with the standard. I would like to discuss two topics here:

In the code described above, does VCS or Verilator better conform to the Verilog standard?
If Verilator does not fully comply with the standard, what methods can be used to identify these potential issues to avoid drawing incorrect conclusions?

jcteng, 2025-04-27:
@wsnyder  any suggestion for this?

FanShupei, 2025-04-27:
Verilator treats code more like a synthesis tool. If the code contains major code style problem (typically introducing synthesis/simulation mismatch), a mismatch between Verilator and other tools following the standard faithfully is expected. Usually I use a lint tool to detect such issues.

In theory, it's better that Verilator could catch such synthesis/simulation code style problems and give diagnostics, but currently Verilator is not good at it.

Disclaimer: It's my personal view as an experienced Verilator user. A developer may give a more relevant comment.

jcteng, 2025-04-27:
@FanShupei Thanks for your hint.  ShuPei

wsnyder, 2025-04-27:
Verilator is not IEEE compliant in all scheduling respects, as @FanShupei mentioned it is more like synthesis.

Also note even if compliant simulators have a lot of latitude as to how they schedule processes, if your code has races you can get differences between simulators.

Anyhow this sort of divergence is not something we intend to "fix" as it's necessary for decent performance.  Ideally we would have a warning, a contribution to do that would be welcome, but not likely to happen soon without a contribution, so closing.

closed 2025-04-27 by wsnyder

jcteng, 2025-04-27:
@wsnyder  Is it possible to generate race warnings for those possible situations (perhaps only some but not all) through some detection conditions?

I would like to hear suggestions in this regard. If there are some algorithms or preliminary ideas that can detect these potential situations at the AST level, we can add implementations to generate relevant warnings, so that developers or AI can adapt the RTL implementation to the current scheduling mode as much as possible?
Another question is, have there been any related discussions before that describe the details of the scheduling differences?

wsnyder, 2025-04-27:
For the case you included it could warn if a variable is set in a process with blocking assignment after it is used.  Most cases are much harder to detect, e.g. if there was a separate "always" statement connecting the first and second usage, although in that case Verilator will usually give an UNOPTFLAT warning.

As to docs, most relevant is probably https://verilator.org/guide/latest/languages.html#synthesis-subset

as always pulls are also welcome to improve the documentation.

jcteng, 2025-04-27:
@wsnyder thanks for your info.