verilator/verilator #5962: Potential bug in evaluating combinatorial wrapping to 0

Cpt-Quantum, 2025-04-27:
Hi, I think I may have encountered a bug in Verilator version 5.034. I'm using cocotb to invoke Verilator for simulating a FIFO.

I'm checking the FIFO is full by matching the write pointer address against the read pointer address as:
`assign FIFO_FULL  = (((wr_adr + 1)) == rd_adr);`
For this case FIFO full doesn't assert even when the wr_adr value reaches 511 (where wr_adr + 1 would hit 512 and wrap to equal 0). Instead to force verilator to reach the correct value I have to wrap the addition with a modulo operator (and add a lint off rule):
`assign FIFO_FULL  = (((wr_adr + 1)%512) == rd_adr); /* verilator lint_off WIDTHEXPAND */`

It also works on the synchronous logic for wr_adr if it's incremented past 511. So I suspect this is just a width check issue on the combinatorial side.

I've attached the full testbench for more detail.
[verilator_test.zip](https://github.com/user-attachments/files/19929707/verilator_test.zip)

FanShupei, 2025-04-27:
`1` is an unsized constant number, and according to LRM it's at least 32-bit, so the addition and the comparison performs in 32-bit. (I believe this is a pretty common footgun of systemverilog) So actually here verilator behaves correctly?

wsnyder, 2025-04-27:
As @FanShupei mentioned that's correct per IEEE, and the == upsizes to largest e.g. 32 bits.  Verilator warned about this.

`%Warning-WIDTHEXPAND: fifo.v:43:43: Operator EQ expects 32 or 10 bits on the RHS, but RHS's VARREF 'rd_adr' generates 9 bit$
 `

closed 2025-04-27 by wsnyder