verilator/verilator #5963: Mismatch when connecting unsigned module output to signed wire in Verilator simulation

sdjasj, 2025-04-27:
Below is the problematic source file **`test.v`**; notice that **`out0` is declared as *unsigned***. I have already reduced the code to the smallest failing example:

```verilog
`timescale 1ns/1ps
module ircxoylsdv (out0 );
    reg signed [27:20] reg_4;
    output wire [21:10] out0;

    initial begin
        #1;
        reg_4 = 0;
    end

    wire [11:0] _055_;
    wire [11:0] _023_;
    assign _055_ = ~reg_4[20];
    assign { _023_[3], _023_[1:0] } = 3'h0;
    assign { _023_[11:4], _023_[2] } = { _055_[11:4], _055_[2] };
    assign out0 = _023_;
endmodule
```

The corresponding testbench **`tb.v`** is shown below; here **`out0` is declared as *signed***:

```verilog
`timescale 1ns/1ps
module tb_dut_module;
    wire signed [21:10] out0;

    ircxoylsdv uut (
        .out0(out0)
    );

    initial begin
        #2000;
        $display("out0: %0d\n", out0);
        $finish;
    end
endmodule
```

I run the simulation with:

```bash
verilator --binary -Wno-lint --timing test.v tb.v
./obj_dir/Vtest
```

and obtain the following output:

```
out0: 4084
```

Observe that **`4084` equals the binary pattern `12'b1111_1111_0100`**, which indicates that **`out0` is being printed as an *unsigned* value inside `tb.v`**.  
The correct signed interpretation of `12'b1111_1111_0100` should be **`-12`**.

**System details**

- Verilator version: **5.035 devel rev UNKNOWN.REV**  
- OS: **Ubuntu 22.04.5 LTS (Jammy)**  
- Kernel/Release: **22.04**

wsnyder, 2025-04-27:
`-fno-expand` is a workaround, Verilator is constant propagating the negative number into a constant string and gets messed up because V3Expand changed the width().  Need to think how to fix properly.

wsnyder, 2025-04-28:
BTW although I haven't finished auditing, I believe the only thing this will affect is $display and similar (e.g. $sformatf), if you have a numerical mistake that is similar but doesn't use any of those, let me know by posting another example here.

closed 2025-05-05 by wsnyder