verilator/verilator #5966: Incorrect UNOPTFLAT lint warning

BertVerrycken, 2025-04-28:
## Description

While trying to design an automatic adder tree based on a parameter I noticed two things:
- The lint check fails with a UNOPTFLAT warning IMO mistaken signal unoptimizable, it sees a loop from input to output back to input of the same signal but in fact it is correct, it is not a loop. Suppressing the warning simulates fine in verilator. Questa has no issue with this code and correctly compiles, elaborates and simulates this.
- While localparam can be different for every generate loop, I didn't find a way to correctly size the internal signals to connect the tree, therefor, all internal signals are defined as [C_NOI/2-1:0], this means that the further we go down the tree, the more unused signals there are (C_NUM_ADDERS is shifted right so less and less signals actually used for increasing level). But there is no unused bits warning from the verilator lint check.

Both lint checks, circular connection (wrong) and no unused bits warning seem to warrant a fix (only if I'm right of course).

## Code

This is not the actual code, I have instances in the tree in the real code so I replaced the instances with a simple unsigned addition but the error stays the same (and absence of unused bits warning):

```systemverilog
package float_pkg;
  typedef logic unsigned [31:0] word32_u_t;
endpackage

module example
  import float_pkg::*;
#(
  parameter int                     C_NOI             = 4)
  (
  input  word32_u_t     [C_NOI-1:0] operand_a,
  input  word32_u_t     [C_NOI-1:0] operand_b,
  output word32_u_t                 result
  );

  localparam int                    C_ALVLS     = $clog2(C_NOI);                   // Number of adder levels

  word32_u_t     [C_NOI/2-1:0]      input_add_a[C_ALVLS];
  word32_u_t     [C_NOI/2-1:0]      input_add_b[C_ALVLS];
  word32_u_t     [C_NOI/2-1:0]      result_add[C_ALVLS];
  generate
    for (genvar lvl = 0; lvl < C_ALVLS; lvl++) begin: add_levels
      // Number of adders at this level
      localparam int C_NUM_ADDERS = C_NOI/2 >> lvl;

      for (genvar i = 0; i < C_NUM_ADDERS; i++) begin: adders
        assign result_add[lvl][i] = input_add_a[lvl][i] + input_add_b[lvl][i];
      end
    end
  endgenerate

  generate
    for (genvar lvl = 0; lvl < C_ALVLS; lvl++) begin: signal_levels
      localparam int C_NUM_ADDERS = C_NOI >> lvl;

      for (genvar i = 0; i < C_NUM_ADDERS; i++) begin: signals
        if (lvl == 0) begin: add_assign_pairs_lvl0
          assign input_add_a[0][i]   = operand_a[i];
          assign input_add_b[0][i]   = operand_b[i];
        end else begin: add_assign_pairs_lvl1
          assign input_add_a[lvl][i] = result_add[lvl-1][i*2];
          assign input_add_b[lvl][i] = result_add[lvl-1][i*2+1];
        end
      end
    end
  endgenerate

  assign result = result_add[C_ALVLS-1][0];

endmodule
``` 

A simple TB:

```systemverilog
module tb;

  // ########################################
  // ## Declarations
  // ########################################
  typedef logic unsigned [31:0] word32_u_t;

  word32_u_t     [3:0] operand_a;
  word32_u_t     [3:0] operand_b;
  word32_u_t           result;

  // ########################################
  // ## DUT
  // ########################################
  example i_example (.*);

  // ########################################
  // ## Test
  // ########################################
  always begin

    for (int i = 0; i < 4; i++) begin
      operand_a[i] = 1;
      operand_b[i] = 1;
    end
    #1;
    $display("result: %d", result);

    #1000;
    $finish;
  end

endmodule
``` 
Since inputs are all 1, the number of instances (C_NOI) is 4, the result is correctly reported as 4 by both verilator and questa.

## Verilator command

Verilator is run via fusesoc.

Three files if you count the package included in the first code section:
- float_pkg.sv
- example.sv
- tb.sv

Sim command:
```
verilator -f core_rtl_hdlcommon_example_1.0.0.vc -Wno-lint -Wno-TIMESCALEMOD --assert --binary -j 0
``` 
Lint command:
```
verilator --lint-only -f core_rtl_hdlcommon_example_1.0.0.vc -Wall --unroll-count 72
```

Replace the fusesoc vc file with the actual source files.

## Tool and OS

- Ubuntu 24.04 LTS on Intel based Mac Mini 2012.
- Verilator v5.034.
- Questa reference sim version = 2023.3.

## May we assist you in trying to fix this in Verilator yourself?

I'm mostly a hardware smurf so I have bad sw skills, but if you think I can help fix this, let me know.

wsnyder, 2025-04-28:
These two issues seem unrelated, so generally we'd like two separate bugs so can reply/track separately.

But anyhow thanks for the good writeup.  It's not clear:

1. What are you suggesting change with UNOPTFLAT given how it is documented to work?  https://verilator.org/warn/UNOPTFLAT

2. UNUSED* warnings don't currently support anything but simple 1D packed arrays, e.g. not unpacked arrays.  We can use this bug as the request to support analyzing them.

wsnyder, 2025-04-29:
Ok, using this to track the UNDRIVEN request.  Note also #850 which is a different UNDRIVEN request for members.  Pull requests to enhance this welcome.