verilator/verilator #5968: Unexpected case matching behavior for signed values compared against unsigned constants

sdjasj, 2025-04-29:
The following is the problematic Verilog code `test.v`:

```verilog
`timescale 1ns/1ps
module a();
reg signed [2:0] in0;

initial begin
  in0 = 7; // signed 3'b111
  #1;
  $display("in0: %0d", in0);
  case (in0)
    // compare signed 3'b111 and unsigned 4'b0111
    // signed 3'b111 ----> unsigned 4'b0111
    // so 4'b0111 == 4'b0111, match
    4'b111: begin
      $display("1");
    end
    default: begin
      $display("2");
    end
  endcase

end

endmodule
```

You can reproduce the issue by running the following commands:

```
verilator --binary -Wno-lint --timing -fno-expand test.v
./obj_dir/Vtest
```

Verilator outputs:

```
in0: -1
2
```

I consulted the Verilog standard **IEEE Std 1364™-2005** and found the following statement:

> When one or both operands of a relational expression are unsigned, the expression shall be interpreted as a comparison between unsigned values. If the operands are of unequal bit lengths, the smaller operand shall be zero-extended to the size of the larger operand.

Therefore, when the value `3'b111` of `in0` is compared with the unsigned constant `4'b111`, `in0` should be treated as an unsigned value. Since `in0` has fewer bits than 4, it should be **zero-extended** to `4'b0111`. As a result, it should match the `4'b111` branch, and `1` should be printed instead of `2`.

System details

* Verilator version: 5.035 devel rev UNKNOWN.REV
* OS: Ubuntu 22.04.5 LTS (Jammy)
* Kernel/Release: 22.04

wsnyder, 2025-04-29:
Every simulator agrees it's -1, please try on other simulators before filing ;)

sdjasj, 2025-04-29:
> Every simulator agrees it's -1
是的，它们的确都输出`in0: -1`，但我想说的是，正常匹配的分支应该是`3'b111`这个分支（匹配时进入分支会输出`1`），而不是`defalut`这个默认的分支（匹配时进入分支会输出`2`），cxxrtl和iverilog这两个开源仿真器都进入的是`3'b111`这个分支，但是verilator进入了`defalut`这个分支

wsnyder, 2025-04-29:
Ah, ok, sorry.  To avoid confusion in tests, please don't rely on $display results, compare values and $stop if the value is not what is expected.  Ideally use the test_regress `checkd macros.  Example:

```
`define stop $stop
`define checkd(gotv,expv) do if ((gotv) !== (expv)) begin $write("%%Error: %s:%0d:  got=%0d exp=%0d\n", `__FILE__,`__LINE__, (gotv), (expv)); `stop; end while(0);

module t();
   reg signed [2:0] in0;
   int match;
   bit cmp;

   initial begin
      in0 = 7; // signed 3'b111

      cmp = in0 == -1;
      `checkd(cmp, 1);
      cmp = in0 != -1;
      `checkd(cmp, 0);

      // verilator lint_off WIDTHEXPAND
      case (in0)
        // compare signed 3'b111 and unsigned 4'b0111
        // signed 3'b111 ----> unsigned 4'b0111
        // so 4'b0111 == 4'b0111, match
        4'b111: begin
           match = 1;
        end
        default: begin
           match = 2;
        end
      endcase
      // verilator lint_on WIDTHEXPAND

      `checkd(match, 1);

      $write("*-* All Finished *-*\n");
      $finish;
   end

endmodule
```

sdjasj, 2025-04-29:
Thank you for your guidance! I didn't know about this method before, but I will definitely use it for debugging in the future. :)

wsnyder, 2025-04-29:
Fixed, there was a test case for this but it had wrong expectation, not sure how that got in.

BTW generally turning off WIDTH warnings using signed numbers is likely to hit bugs.

closed 2025-04-29 by wsnyder

sdjasj, 2025-04-29:
Thank you for the quick fix! By the way, you are one of the most efficient developers I’ve ever seen. :)