verilator/verilator #5969: Verilator Internal Error with 4-state Variable During Simulation Using -fno-expand Option

sdjasj, 2025-04-30:
Hello! Recently, I've been using Verilator to simulate a Verilog file optimized by Yosys, and I inadvertently encountered the following issue.

Here is the optimized code file, which contains several thousand lines of simple Verilog statements. I tried minimizing it, but the results were not ideal.

[[opt.txt](https://github.com/user-attachments/files/19969601/opt.txt)]

This is the corresponding tb file.

[[tb.txt](https://github.com/user-attachments/files/19969675/tb.txt)]

You can reproduce the error using the following command:

```
verilator --binary -Wno-lint --timing -fno-expand opt.txt tb.txt
```

The Verilator output is as follows:

```
%Error: Internal Error: opt.v:2115:24: ../V3Number.cpp:975: toUQuad with 4-state 32'bxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx0
 2115 |   assign out51 = _0539_[0];
      |                        ^
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
```

Although I couldn't minimize the issue, I found some interesting observations during debugging. I discovered that I need to compile both `tb.v` and `test.v` together and enable the `-fno-expand` option to trigger this error; otherwise, it can compile normally.

I'm not sure if this behavior is expected. Since the file contains several thousand lines of code, it may be difficult for developers to pinpoint and fix the issue immediately. However, I believe it would still be valuable to share this information and make others aware of the problem. :)

System details:
* Verilator version: 5.035 devel rev UNKNOWN.REV  
* OS: Ubuntu 22.04.5 LTS (Jammy)  
* Kernel/Release: 22.04

wsnyder, 2025-04-30:
Please use sv-bugpoint to reduce it further.  It's obvious that given there are "(*...*)" attributes that could be removed that this can be much smaller with a small effort.

sdjasj, 2025-04-30:
> Please use sv-bugpoint to reduce it further.

Okay, I previously used creduce to minimize, but the results were not good. Now I'm trying to use sv-bugpoint.

sdjasj, 2025-04-30:
I used sv-bugpoint for minimization, and it was indeed more useful than creduce—thank you for the guidance. Below is the minimized code `test.v` after using sv-bugpoint and some manual reduction:

```verilog
`timescale 1ns/1ps
module egakzxppxv(in0, in2, clock_1, out51);
  wire _0090_;
  reg _0200_ = 1;
  reg [10:0] _0216_ = 1;
  wire [11:0] _0258_;
  reg [22:0] _0274_ = 1;
  wire [16:0] _0539_;
  input clock_1;
  input [23:11] in0;
  input [27:8] in2;
  wire [21:5] out21;
  output out51;
  reg [24:15] reg_18 = 10'h000;
  assign _0104_ = ! _0509_;
  assign _0201_ = _0258_ > 12'hab7;
  assign _0204_ = 7'h7f <= _0104_;
  assign _0258_ = 3'h3 ||  _0274_;
  always @(negedge clock_1, posedge in0)
    if (_0090_) reg_18 <= 10'h09d;
  assign _0412_ = | reg_18[19:18];
  assign _0413_ = | _0537_;
  assign _0537_ = in2[11] ?  11'h03e : _0216_;
  assign _0538_ = _0412_ ?  out21 : _0201_;
  assign _0539_ = _0413_ ?  _0538_ : 17'h00570;
  assign out21 = _0204_;
  assign out51 = _0539_[0];
  assign _0509_ = _0200_;
endmodule
```

Running the following command causes an error:
```
verilator --binary -Wno-lint --timing -fno-expand test.v
```

Verilator output:
```
%Error: Internal Error: opt2.v:24:26: ../V3Number.cpp:975: toUQuad with 4-state 32'bxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx0
   24 |   assign _0538_ = _0412_ ?  out21 : _0201_;
      |                          ^
                        ... See the manual at https://verilator.org/verilator_doc.html?v=5.037 for more assistance.
```

However, running the following command does **not** cause an error:
```
verilator --binary -Wno-lint --timing test.v
```

wsnyder, 2025-04-30:
Thanks.  I understand what's going on, it's similar-ish to #5963, I need to think about if there's a general fix to solve this whole category.  A simple experiment suggested it's not so simple ;)

sdjasj, 2025-04-30:
> Thanks. I understand what's going on

Glad to help confirm the issue :)

wsnyder, 2025-04-30:
Fixed similar to some other cases that were special cased.  I'll leave the more general fixing as part of considering #5963.

closed 2025-04-30 by wsnyder