verilator/verilator #5971: $finish doesn't stop the simulation immediately when called inside `always` block

Shatur, 2025-05-01:
`verilator --version`: Verilator 5.036 2025-04-27 rev v5.036
OS: ArchLinux

I have the following code:
```sv
module tb2 ();
    parameter CLK_PERIOD = 2;

    reg clk = 1'b0;

    always #(CLK_PERIOD / 2) clk = ~clk;

    always begin
        int counter = 0;
        while (counter < 3) begin
            counter += 1;
            $display("running loop %d", counter);
            @(posedge clk);
        end

        $finish;
    end
endmodule
```

I would expect the following output:
```
running loop           1
running loop           2
running loop           3
- tb2.sv:16: Verilog $finish
```

But the actual output is this:
```
running loop           1
running loop           2
running loop           3
- tb2.sv:16: Verilog $finish
running loop           1
```

We tested the exact same testbench on Xsim and it works as expected.

wsnyder, 2025-05-01:
For legacy/speed/simplicity Verilator doesn't use exceptions or otherwise to check finish at every statement.  It instead has checks at end of each timestep and inside loops.  Probably something with the timing implementation is missing a check.   Might you look into making a pull to fix this?

wsnyder, 2025-05-01:
I'm not sure why you also filed #5970 but same thing.

Shatur, 2025-05-01:
> Might you look into making a pull to fix this?

I'm not sure if I have enough knowledge to fix it myself 😢 

> I'm not sure why you also filed https://github.com/verilator/verilator/issues/5970 but same thing.

I didn't know it's the same thing... In this case it just runs one extra time. But in #5970 it runs infinitely. I assumed it's 2 different issues.

wsnyder, 2025-05-02:
Appreciate the good test case, fixed.

Note that in general for performance reasons, some same-cycle process events will still run after $finish, but none should continue past the current time step (like this one did).

closed 2025-05-02 by wsnyder