verilator/verilator #5972: Incorrect assignment behavior with `-fno-expand`

sdjasj, 2025-05-01:
The following Verilog code in `test.v` exhibits an issue:

```verilog
`timescale 1ns/1ps
`define stop $stop
`define checkd(gotv,expv) do if ((gotv) !== (expv)) begin $write("%%Error: %s:%0d:  got=%0d exp=%0d\n", `__FILE__,`__LINE__, (gotv), (expv)); `stop; end while(0);

module out24_logic_simplify (
    input  wire               clock_0,
    input  wire               clock_1,
    input  wire signed [28:28] in1,
    output wire        [21:5] out24
);
    reg signed [21:8] reg_10;

    initial begin
        reg_10 = 0;
    end

    assign out24 = reg_10;

    always @(negedge clock_1 or posedge clock_0) begin
        if (clock_0) begin
            reg_10 <= 0;
        end else begin
            reg_10[14:8] <= {1'b1, ~((in1[28:28] & ~(in1[28:28])))};  // Should assign 3
        end
    end
endmodule

module tb_out24_logic_print;

    reg  clock_0;
    reg  clock_1;
    reg  signed [28:28] in1;
    wire [21:5] out24;
    bit cmp;

    out24_logic_simplify dut (
        .clock_0 (clock_0),
        .clock_1 (clock_1),
        .in1     (in1),
        .out24   (out24)
    );

    initial begin
        clock_0 = 0;
        clock_1 = 1;
        in1     = 1'b0;
        cmp = out24 == 0;
        `checkd(cmp, 1);
        #2;
        clock_1 = 0;
        #2;
        cmp = out24 == 1; // line 54
        `checkd(cmp, 1);
        cmp = out24 == 3;
        `checkd(cmp, 1); // line 57
        $finish;
    end
endmodule
```

This issue can be reproduced by running (removing `-fno-expand` prevents the incorrect simulation result.):

```bash
verilator --binary -Wno-lint --timing -fno-expand test.v
```

Verilator then outputs the following error:

```
%Error: test.v:57:  got=0 exp=1
%Error: test.v:57: Verilog $stop
Aborting...
Aborted (core dumped)
```

As shown, the check at line 57 (`cmp = out24 == 3`) fails. However, based on the logic in `out24_logic_simplify`, when `reg_10[14:8] <= {1'b1, ~((in1[28:28] & ~(in1[28:28])))};` is triggered, `reg_10` should be updated such that `out24 == 3`. But from line 54, we can see that Verilator incorrectly gives a result of `1`.

wsnyder, 2025-05-01:
Verilator seems to give same results as other simulators, please check.

sdjasj, 2025-05-01:
> Verilator seems to give same results as other simulators, please check.

Sorry, the issue disappeared after I updated Verilator to the latest version. Apologies for wasting your time.

sdjasj, 2025-05-01:
Here is the system information I used:
* Verilator 5.037 devel rev v5.036-18-g2c0372acd (mod)
* OS: Ubuntu 22.04.5 LTS (Jammy) Kernel/Release: 22.04

wsnyder, 2025-05-02:
Still fails on all commercial sims & verilator.

sdjasj, 2025-05-02:
> Still fails on all commercial sims & Verilator.

Can you try the following code?

```verilog
`timescale 1ns/1ps

module out24_logic_simplify (
    input  wire               clock_0,
    input  wire               clock_1,
    input  wire signed [28:28] in1,
    output wire        [21:5] out24
);
    reg signed [21:8] reg_10;

    initial begin
        reg_10 = 0;
    end

    assign out24 = reg_10;

    always @(negedge clock_1 or posedge clock_0) begin
        if (clock_0) begin
            reg_10 <= 0;
        end else begin
            reg_10[14:8] <= {1'b1, ~((in1[28:28] & ~(in1[28:28])))}; 
        end
    end
endmodule

module tb_out24_logic_print;

    reg  clock_0;
    reg  clock_1;
    reg  signed [28:28] in1;
    wire [21:5] out24;
    bit cmp;

    out24_logic_simplify dut (
        .clock_0 (clock_0),
        .clock_1 (clock_1),
        .in1     (in1),
        .out24   (out24)
    );

    initial begin

        clock_0 = 0;
        clock_1 = 1;
        in1     = 1'b0;
        #2;
        clock_1 = 0;
        #2;
        $display("out24: %0d", out24);
        $finish;
    end
endmodule
```

Compile it using the following command:

```
verilator --binary -Wno-lint --timing -fno-expand test.v
```

Then run:

```
./obj_dir/Vtest
```

Verilator will print:

```
out24: 1
```

However, according to the following line of code, the result shouldn't be `out24: 1`, because the expression `{1'b1, xxxx}` guarantees the lowest 2 bits are set to `11`, which means the output should **not** be 1. I ran the same test with Icarus Verilog and CXXRTL, and both printed `out24: 3`.

```verilog
reg_10[14:8] <= {1'b1, ~((in1[28:28] & ~(in1[28:28])))}; 
```

wsnyder, 2025-05-02:
Same root case as #5963 but good to have this case to make sure that wider fix does this one too.

sdjasj, 2025-05-02:
> Same root case as [#5963](https://github.com/verilator/verilator/issues/5963) but good to have this case to make sure that wider fix does this one too.

Thank you for taking the time to analyze this issue! I’d like to ask if there’s any way to avoid triggering this bug. I'm running some tests on Verilator and would like to avoid repeatedly triggering the same error.

wsnyder, 2025-05-02:
Is there a reason you are using -fno-expand, or does your real case not use that?

Generally the problem is with constant propagation very late in processing, most designs don't have so much tied-off logic which gets constant-propagated, which is why we're seeing these issues for the first time.

sdjasj, 2025-05-02:
> Is there a reason you are using -fno-expand, or does your real case not use that?

Haha, that's because I want to avoid the bug in [https://github.com/verilator/verilator/issues/5963](https://github.com/verilator/verilator/issues/5963). And at the time, your suggestion to me was to use `-fno-expand` :).

wsnyder, 2025-05-02:
Try `-fno-subst -fno-subst-const`

sdjasj, 2025-05-02:
> -fno-subst -fno-subst-const

Thank you for your patient reply. I'll give it a try.