verilator/verilator #5974: Assign "null" to virtual interface

fonarick, 2025-05-03:
Hello, verilator throws an error when i try to assign "null" to virtual interface
```systemverilog
interface test_if;
endinterface

module top();
	test_if real_if();
	virtual test_if virt_if;
	initial begin
		virt_if = real_if;
		virt_if = null;
	end
endmodule
``` 
its allowed by standard (IEEE Std 1800-2012 - Interface - Virtual interface)
```
Only the following operations are directly allowed on virtual interface variables:
— Assignment ( = ) to the following:
    — Another virtual interface of the same type
    — An interface instance of the same type
    — The special constant null                              <---- this one
— Equality ( == ) and inequality ( != ) with the following:
    — Another virtual interface of the same type
    — An interface instance of the same type
    — The special constant null
``` 
and simulation starts in vivado

script:
```
verilator --binary -j 0 --threads "$(nproc)" --Mdir build --top-module top top.sv
```
log:
```
%Error: top.sv:9:13: Operator ASSIGN expected 'test_if' interface on Assign RHS but ''0' is not an interface.
                   : ... note: In instance 'top'
    9 |   virt_if = null;
      |             ^~~~
        ... See the manual at https://verilator.org/verilator_doc.html?v=5.035 for more assistance.
%Error: Exiting due to 1 error(s)
``` 
version: ```Verilator 5.035 devel rev v5.034-182-gc9be36911``` 
Thanks

fonarick, 2025-05-03:
btw inequality (!=) comparison with null not compiling
```systemverilog
interface test_if;
endinterface

module top();
	test_if real_if();
	virtual test_if virt_if;
	initial begin
		virt_if = real_if;
		if (virt_if != null)
			$display("real");
		else
			$display("virt");
	end
endmodule
``` 
short log:
```
In file included from Vtop__ALL.cpp:9:
Vtop___024root__DepSet_h84412442__0__Slow.cpp: In function ‘void Vtop___024root___eval_initial__TOP(Vtop___024root*)’:
Vtop___024root__DepSet_h84412442__0__Slow.cpp:18:19: error: no match for ‘operator!=’ (operand types are ‘VlNull’ and ‘Vtop_test_if*’)
   18 |     if ((VlNull{} != top__DOT__virt_if)) {
      |          ~~~~~~~~ ^~ ~~~~~~~~~~~~~~~~~
      |          |           |
      |          VlNull      Vtop_test_if*

```

wsnyder, 2025-05-04:
Thanks for the good report.  Might you try a fix for this?  It's using VlNull{} because it thinks it's a class reference, just changing to normal nullptr should work.  In the test be sure to check both == and !=.

fonarick, 2025-05-04:
ok, i'll try 👍