verilator/verilator #5977: Incorrect "syntax error, unexpected IDENTIFIER, expecting IDENTIFIER-for-type"

solomatnikov, 2025-05-05:
Thanks for taking the time to report this.

Can you attach an example that shows the issue?  (Must be openly licensed, completely self-contained and self-checking so can directly run what you provide. Ideally use test_regress format, see https://veripool.org/guide/latest/contributing.html?highlight=test_regress#reporting-bugs

From https://github.com/pulp-platform/axi

```
package axi_test;
  class axi_ax_beat #(
    parameter AW = 32,
    parameter IW = 8 ,
    parameter UW = 1
  );
    rand logic [IW-1:0] ax_id     = '0;
    rand logic [AW-1:0] ax_addr   = '0;
    logic [7:0]         ax_len    = '0;
    logic [2:0]         ax_size   = '0;
    logic [1:0]         ax_burst  = '0;
    logic               ax_lock   = '0;
    logic [3:0]         ax_cache  = '0;
    logic [2:0]         ax_prot   = '0;
    rand logic [3:0]    ax_qos    = '0;
    logic [3:0]         ax_region = '0;
    logic [5:0]         ax_atop   = '0;  
    rand logic [UW-1:0] ax_user   = '0;
  endclass

  class axi_driver #(
    parameter int  AW = 32  ,
    parameter int  DW = 32  ,
    parameter int  IW = 8   ,
    parameter int  UW = 1   ,
    parameter time TA = 0ns ,  
    parameter time TT = 0ns    
  );
    typedef axi_ax_beat #(.AW(AW), .IW(IW), .UW(UW)) ax_beat_t;
  endclass
endpackage

module tb_axi_bus_compare #(
  parameter time TbTclk = 10ns,
  parameter int unsigned TbAddrWidth = 32'd64,
  parameter int unsigned TbDataWidth = 32'd128,
  parameter int unsigned TbIdWidth = 32'd6,
  parameter int unsigned TbUserWidth = 32'd2,
  parameter bit TbWarnUninitialized = 1'b0,
  parameter time TbApplDelay = 2ns,
  parameter time TbAcqDelay = 8ns
);

  typedef axi_test::axi_driver #(
    .AW(TbAddrWidth), .DW(TbDataWidth), .IW(TbIdWidth), .UW(TbUserWidth),
    .TA(1ns), .TT(6ns)
  ) drv_t;

  localparam int StrbWidth = 1;

  initial begin
    automatic logic rand_success;
    automatic drv_t::ax_beat_t aw_beat = new, ar_beat = new;

    rand_success = aw_beat.randomize(); assert (rand_success);
    aw_beat.ax_addr >>= $clog2(StrbWidth);  
    aw_beat.ax_addr <<= $clog2(StrbWidth);

     $finish();
  end
endmodule
```

VCS compiles w/o problem.

What 'verilator' command line do we use to run your example?

```
verilator --binary test.sv
```

What 'verilator --version' are you using?  Did you try it with the git master version?

```
Verilator 5.037 devel rev v5.036-30-g11cfa61f8
```

What OS and distribution are you using?

```
Linux 5.14.0-503.38.1.el9_5.x86_64
```

May we assist you in trying to fix this in Verilator yourself?

(Please avoid attaching screenshots that show text - you can convert images to text using e.g. https://ocr.space)

wsnyder, 2025-05-20:
No longer blocked by #2412.  Issue is now that typedef which is of a parameterized class gets resolved before the class is parameterized.  This is a workaround:

```
-    automatic drv_t::ax_beat_t aw_beat = new;
+    automatic Pkg::axi_driver #(.AW(TbAddrWidth)) aw_beat = new;
```

solomatnikov, 2025-05-21:
Will this be fixed?

It would be hard for such "workaround" to get accepted/merged.

wsnyder, 2025-05-21:
Will you contribute a pull?   More seriously, yes, eventually, but we have many issues, but not enough time ;)

solomatnikov, 2025-06-02:
> Will you contribute a pull? 

I wish I knew how to do it but unfortunately my understanding of Verilator internals is not sufficient and from your earlier comments it sounds like this is related to a recent complicated redesign.

> More seriously, yes, eventually, but we have many issues, but not enough time ;)

wsnyder, 2025-11-17:
Another smaller example:

```
package uvm_pkg;
  class uvm_reg_sequence #(
      type BASE = int
  );
    typedef enum {
      LOCAL,
      UPSTREAM
    } seq_parent_e;
  endclass
endpackage

module t;
  import uvm_pkg::*;
  class test_seq extends uvm_reg_sequence;
  endclass
  initial begin
    test_seq c;
    c = new;
  end
endmodule
```

closed 2026-03-25 by wsnyder