verilator/verilator #5984: --trace-fst causes internal error on unpacked array with shifted index assignment

hankhsu1996, 2025-05-05:
When using `--trace-fst`, Verilator hits an internal error if you assign to an unpacked array with shifted indices. Without `--trace-fst`, everything works fine. 

## Code Snippet

```systemverilog
module Test #(
  localparam int DIM1 = 4,
  localparam int DIM2 = 8,
  localparam int SHIFT = 2
) (
  input logic [DIM1-1:0][DIM2-1:0] in
);

  logic [DIM1-1:0][DIM2+SHIFT-1:SHIFT] internal;

  for (genvar dim1 = 0; dim1 < DIM1; dim1++) begin
    for (genvar dim2 = 0; dim2 < DIM2; dim2++) begin
      assign internal[dim1][dim2+SHIFT] = in[dim1][dim2];
    end
  end

  for (genvar dim1 = 0; dim1 < DIM1; dim1++) begin
    for (genvar dim2 = 0; dim2 < DIM2; dim2++) begin
      always_ff @(negedge internal[dim1][dim2+SHIFT]) begin
        $display("%0b", internal[dim1][dim2+SHIFT]);
      end
    end
  end

endmodule
```

Use this command:

```bash
verilator test.sv --trace-fst --cc --exe --binary --main --top Test
```

It only crashes if `--trace-fst` is used. Without it, no error.

Here’s the error Verilator throws:

```
%Error: Internal Error: test.sv:9:40: ../V3FuncOpt.cpp:228: Inconsistent assignment
   9 |   logic [DIM1-1:0][DIM2+SHIFT-1:SHIFT] internal;
      |                                        ^~~~~~~~
```

## Version

Verilator 5.036 (2025-04-27)
Ubuntu 22.04
Always reproducible with `--trace-fst`

hankhsu1996, 2025-05-05:
Just want to mention: I was previously using v5.028, and everything worked fine. After upgrading to v5.036, I encountered several new issues:
- Interface hierarchical access (already reported in #5941)
- This internal error
- ../V3SplitVar.cpp:1153: not enough split variables: This one also happens when --trace-fst is on. Maybe it's related to this issue, but it's hard to reproduce with a small test case.

wsnyder, 2025-05-05:
`-fno-var-split` is a workaround, it seems to get upset by the multidimensional arrays and splits incorrectly.

@yTakatsukasa might you take a look?   Also note to @gezalore, this seems a fallout of splitting by default, but I presume the base issue was there earlier.

Simplified test case:

```
module t (
  input logic [1:0][3:0] in
  );

  logic [1:0][5:2] internal;

  for (genvar dim1 = 0; dim1 < 2; dim1++) begin
    for (genvar dim2 = 0; dim2 < 4; dim2++) begin
      assign internal[dim1][dim2+2] = in[dim1][dim2];
    end
  end

  for (genvar dim1 = 0; dim1 < 2; dim1++) begin
    for (genvar dim2 = 0; dim2 < 4; dim2++) begin
      always_ff @(negedge internal[dim1][dim2+2]) begin
        $display("%0b", internal[dim1][dim2+2]);
      end
    end
  end

endmodule
```

wsnyder, 2025-05-05:
Also note `trace-fst` is needed only to prevent the original signal rebuilt from components from getting optimized away.

hankhsu1996, 2025-05-05:
Add another issue 😛 

```
ver___024root__DepSet_h96c49fed__23.cpp: In function 'void ver___024root___eval_triggers__act__14(ver___024root*)':
ver___024root__DepSet_h96c49fed__23.cpp:4753:46: error: '__Vtrigcurrexpr_h1cad7197__0' was not declared in this scope
 4753 |     vlSelfRef.__Vtrigprevexpr_h1cad7197__0 = __Vtrigcurrexpr_h1cad7197__0;
      |                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

Looks like the SenExprBuilder is emitted to different cpp files. I'm almost certain this breaks after #5798. Maybe it's the same issue?

wsnyder, 2025-05-05:
That's certainly a separate issue from the index assignment one, please file a separate issue, thanks.

PR #5988 (2025-05-06): Fix Inconsistent assignment error by split_var
Fixes #5984

I suspect #5785 caused this issue.

As usual, run a test to reproduce the issue, then push the fix.
closed 2025-05-06 by wsnyder