verilator/verilator #5989: Node has no type Error on cast within concatenation

obliguedon, 2025-05-06:
Thanks for taking the time to report this.

Can you attach an example that shows the issue?  (Must be openly licensed, completely self-contained and self-checking so can directly run what you provide. Ideally use test_regress format, see https://veripool.org/guide/latest/contributing.html?highlight=test_regress#reporting-bugs
-`test.sv`:
```verilog
package my_pkg;
  typedef enum logic [1:0] {
    SIG_0, SIG_1, SIG_2
  } sig_t;
endpackage : my_pkg

module test
  import my_pkg::*;
(
  input logic [6:0] data,
  output logic [2:0] opt
);
  assign opt = {data[5], sig_t'(data[1:0])};
endmodule : test
```

What 'verilator' command line do we use to run your example?

```shell
verilator --cc --exe --main --Wno-fatal ./test.sv
```

What 'verilator --version' are you using?  Did you try it with the git master version?

```
Verilator 5.037 devel rev v5.036-32-g66e105b44
```

What OS and distribution are you using?

```
OS: CentOS Linux 7
Kernel: Linux 3.10.0-957.5.1.el7.x86_64
```

May we assist you in trying to fix this in Verilator yourself?
- I'm far from good in C++

reported message:
```
%Error: Internal Error: test.sv:16:31: ../V3Width.cpp:7311: Node has no type
                                     : ... note: In instance 'test'
   16 |    assign opt = {data[5], sig_t'(data[1:0])};
      |                                ^
```

obliguedon, 2025-05-06:
it work like this:

```verilog
module test
  import my_pkg::*;
(
input logic [6:0] data,
  output logic [2:0] opt
);
  sig_t sig;
  assign sig = sig_t'(data[1:0]);
  assign opt = {data[5], sig};

endmodule : test
```

but the cast within concatenation work with other simulator

wsnyder, 2025-05-11:
I have a patch to fix it... but it breaks other things, so needs more attention.

toddstrader, 2025-05-16:
> I have a patch to fix it... but it breaks other things, so needs more attention.

@wsnyder  not sure how in-the-weeds the additional attention is, but if it would help I can attempt to push more on this if you can push it to a branch.

wsnyder, 2025-05-16:
The fix for this was just:

```
diff --git a/src/V3Width.cpp b/src/V3Width.cpp
index 9e307391c..562697ba6 100644
--- a/src/V3Width.cpp
+++ b/src/V3Width.cpp
@@ -2100,6 +2100,7 @@ class WidthVisitor final : public VNVisitor {
                                               VFlagChildDType{}, refp};
             nodep->replaceWith(newp);
             VL_DO_DANGLING(pushDeletep(nodep), nodep);
+            userIterate(newp, m_vup);  //<----- added this
         } else {
             nodep->v3warn(E_UNSUPPORTED,
                           "Unsupported: Cast to " << nodep->dtp()->prettyTypeName());

```

Unfortunately the streaming operator tests didn't like that, so that's what needs investigation.

wsnyder, 2025-05-16:
There's an ugly interaction between streaming and cast, because IEEE linked the two operations in a weird way.  Maybe it would be a lot cleaner to earlier move the cast into part of the streaming Ast node.

wsnyder, 2025-05-16:
Yes, I was contemplating that ast edit, but might not help, unsure.  I'm not working on this now so all help welcome.

toddstrader, 2025-05-16:
I updated #6012 with a stab at this.  I'll double-check next week that the PR also handles the code posted in this issue, but I'm pretty sure it will.