verilator/verilator #5991: t_math_signed3 test doesn't work with -O0 option

RRozak, 2025-05-07:
Running that test with the `-O0` option results in the following error:
```
%Error: t/t_math_signed3.v:44:  got='h0 exp='hf
%Error: t/t_math_signed3.v:44: Verilog $stop
Aborting...
```
It is not the only check that fails in this test. The others are in lines 47, 69 and 70.

wsnyder, 2025-05-07:
Thanks for the tests/fix, do you expect to be making pulls for the rest soon-ish, or will this park for a while and be "in the queue"?

RRozak, 2025-05-07:
I expect to work on this in the near future.

RRozak, 2025-05-09:
I found a minimal example:
```systemverilog
module t (/*AUTOARG*/);
   wire [3:0]        bug733_a = 4'b0010;
   wire [3:0]        bug733_yu = $signed(|bug733_a);
   initial begin
      if (bug733_yu != 4'b1111) $stop;
      $finish;
   end
endmodule
```
The command is `verilator --binary <file> -Wno-WIDTH -O0`.
It results in the following cpp code:
```cpp
VL_ATTR_COLD void Vsigned_t___eval_initial__TOP__t(Vsigned_t* vlSelf) {
    VL_DEBUG_IF(VL_DBG_MSGF("+  Vsigned_t___eval_initial__TOP__t\n"); );
    Vsigned__Syms* const __restrict vlSymsp VL_ATTR_UNUSED = vlSelf->vlSymsp;
    auto& vlSelfRef = std::ref(*vlSelf).get();
    // Body
    vlSelfRef.__PVT__bug733_a = 2U;
    if (VL_UNLIKELY(((0xfU != (IData)(vlSelfRef.__PVT__bug733_yu))))) {
        VL_STOP_MT("ex/signed.sv", 5, "");
    }
    VL_FINISH_MT("ex/signed.sv", 6, "");
}

VL_ATTR_COLD void Vsigned_t___stl_sequent__TOP__t__0(Vsigned_t* vlSelf) {
    VL_DEBUG_IF(VL_DBG_MSGF("+  Vsigned_t___stl_sequent__TOP__t__0\n"); );
    Vsigned__Syms* const __restrict vlSymsp VL_ATTR_UNUSED = vlSelf->vlSymsp;
    auto& vlSelfRef = std::ref(*vlSelf).get();
    // Body
    vlSelfRef.__PVT__bug733_yu = (0xfU & VL_EXTENDS_II(4,1, 
                                                       VL_REDOR_I((IData)(vlSelfRef.__PVT__bug733_a))));
}
```
The problem is that the 2nd assignment is not in `eval_initial`.
That assignment also wasn't converted to an `initial` block in contrary to the 1st one, but even if I disable that conversion, the generated cpp files are exactly the same.

wsnyder, 2025-05-09:
Note technically that code has a race.  It should be:

```
module t (/*AUTOARG*/);
   wire [3:0]        bug733_a = 4'b0010;
   wire [3:0]        bug733_yu = $signed(|bug733_a);
   initial begin
      #1;
      if (bug733_yu != 4'b1111) $stop;
      $finish;
   end
endmodule
```

And use --timing.  It's likely other tests have this problem, as most tests predate the timing stuff, they should have use the "cyc" comparisons like in t_EXAMPLE.t (which is another way to fix this test), but if it passed with the simpler form assuming Verilator's usual rules with default `-O` flags it likely was committed..

RRozak, 2025-05-09:
Yes, you're right. I will fix the test.

closed 2025-05-09 by wsnyder