verilator/verilator #5993: Verilator outputs incorrect result for conditional ternary expression involving bit-slice and shift

sdjasj, 2025-05-08:
Here is the problematic code `test.v`:

```verilog
`timescale 1ns/1ps
module top (in3, in4, out10);
    input wire  [7:7] in3;
    input wire [23:16] in4;
    wire  [29:29] wire_2;
    output wire  [28:14] out10;

    assign wire_2 = (((in3) % 4'o16) ? {{4{14'b010111101}}} : (in4[18:18] >> 8'b0001));
    assign out10 = wire_2;
endmodule
```

And the corresponding testbench `tb.v`:

```verilog
`timescale 1ns/1ps

module tb_dut_module;

    reg  [7:7] in3;
    reg signed [23:16] in4;
    wire  [28:14] out10;

    top uut  (
        .in3(in3), .in4(in4), .out10(out10)
    );

    initial begin
        #2000;
        in3 = 0; in4 = 8'b0011_1010; 
        #2000;
        $display("out10: %0d\n", out10);
        $finish;
    end

endmodule
```

The command to reproduce the issue:

```
verilator --binary -Wno-lint --timing test.v tb.v
./obj_dir/Vtest
```

Verilator outputs:

```
out10: 1
```

However, Icarus Verilog outputs:

```
out10: 0
```

And after manual analysis, the correct output should indeed be `0`.
Since `in3 = 0`, the expression:

```verilog
assign wire_2 = (((in3) % 4'o16) ? {{4{14'b010111101}}} : (in4[18:18] >> 8'b0001));
```

simplifies to:

```verilog
assign wire_2 = (in4[2:2] >> 8'b0001);
```

Given that `in4 = 8'b0011_1010`, `in4[2:2] = 0`, so:

```verilog
(in4[2:2] >> 8'b0001) == 0
```

Therefore:

```verilog
out10 = wire_2 = 0
```

System details:

* Verilator version: Verilator 5.037 devel rev v5.036-44-ga80aa07de
* OS: Ubuntu 22.04.5 LTS (Jammy) Kernel/Release: 22.04

wsnyder, 2025-05-08:
@yTakatsukasa seems related to `-fno-const-bit-op-tree`, mind taking a look?

```
`define stop $stop
`define checkd(gotv,expv) do if ((gotv) !== (expv)) begin $write("%%Error: %s:%0d:  got=%0d exp=%0d\n", `__FILE__,`__LINE__, (gotv), (expv)); `stop; end while(0);

module t;

   reg in3;
   reg [23:16] in4;

   // verilator lint_off WIDTH
   wire wire_2 = in3 ? {4{14'b010111101}} : (in4[18] >> 8'b1);
   // verilator lint_on WIDTH

   initial begin
      #10;
      in3 = 0; in4 = 8'b00111__0__10;
      #10;
      `checkd(wire_2, '0);
      #10;
      in3 = 0; in4 = 8'b00111__1__10;
      #10;
      `checkd(wire_2, '0);
      $write("*-* All Finished *-*\n");
      $finish;
   end

endmodule

```

sdjasj, 2025-05-09:
@wsnyder Hi, I'd like to ask whether using `-fno-const-bit-op-tree` can avoid this issue. I think I may have found another problem, but it still gets triggered even with `-fno-const-bit-op-tree`, so maybe it's a different issue?

wsnyder, 2025-05-09:
Probably different.  Why does your code have so many WIDTH problems and constants getting propagated?  Are you generating random test case. If it's "real" code it should be much cleaner.

Not saying these shouldn't be fixed in Verilator, but if the Verilog code was more "normal" it would be much more stable.

sdjasj, 2025-05-09:
> Probably different. Why does your code have so many WIDTH problems and constants getting propagated? Are you generating random test case. If it's "real" code it should be much cleaner.

Hi, yes, you're right—I'm indeed using fuzz testing to generate these Verilog test cases. The code isn't representative of typical hand-written designs; it's intentionally crafted to push boundary conditions and discover potential issues in Verilator. I completely understand that such cases might look unusual, but my aim is precisely to help improve Verilator's robustness against edge cases. Thanks for your patience and the helpful feedback! :)
I hope my bug report didn't cause you any trouble.

wsnyder, 2025-05-09:
Ok, the problem with fuzzing is it makes these cases that while bugs are way outside normal reasonable usage.  I'd request you not have more than one bug open at any given time from fuzzing.  I'm also going to make fixing these lower priority personally, as it it taking time from other issues that are blocking "real user" code.  What would be most helpful to the project is if you would take it a step further and debug and provide fixes along with the issues, please consider.

sdjasj, 2025-05-09:
Got it—totally understand your priorities. I’ll keep only one fuzz issue open at a time and focus more on providing root cause analysis or patches when possible. Thanks for the feedback and all your work on the project!

PR #5998 (2025-05-10): Fix  #5993 by considering intermediate one bit mask
Fixes #5993

Intermediate "& 1" was ignored.

The first commit reproduces the issue like [this](https://github.com/verilator/verilator/actions/runs/14942388583/job/41981369930?pr=5998#step:6:308). Thanks Wilson for the simplified test
closed 2025-05-10 by wsnyder