verilator/verilator #5994: Arithmetic right-shift by constant > 32 bits gives wrong result

sdjasj, 2025-05-09:
The code below triggers the problem:

```verilog
`timescale 1ns/1ps
`define stop   $stop
`define checkd(gotv, expv) \
  do if ((gotv) !== (expv)) begin \
       $write("%%Error: %s:%0d:  got=%0d exp=%0d\n", `__FILE__, `__LINE__, (gotv), (expv)); \
       `stop; \
     end while(0);

module top (out35);
    output wire [2:0] out35;
    wire   signed [2:0] wire_4;
    assign wire_4 = 3'b011;
    assign out35  = (wire_4 >>> 36'hffff_ffff_f);

    initial begin
        #10;
        `checkd(out35, '0);
        $write("*-* All Finished *-*\n");
        $finish;
    end
endmodule
```

Reproduce it with the command below (`-fno-const-bit-op-tree` is optional; removing it makes no difference):

```bash
verilator --binary -Wno-lint --timing -fno-const-bit-op-tree test.v
```

Verilator outputs:

```
%Error: test.v:15:  got=7 exp=0
%Error: test.v:15: Verilog $stop
Aborting...
Aborted (core dumped)
```

Because the sign-bit of `wire_4` is `0`, the final result should be `0`. Icarus Verilog (iverilog) confirms this:

```
*-* All Finished *-*
```

Note: the error only occurs when the arithmetic right-shift amount for `wire_4` is greater than 32 bits (`32'hffff_ffff`). The issue is likely related to that.

System details:

* Verilator version: Verilator 5.037 devel rev v5.036-44-ga80aa07de
* OS: Ubuntu 22.04.5 LTS (Jammy) Kernel/Release: 22.04

sdjasj, 2025-05-09:
During testing, I noticed that sometimes Verilator reports an error like the following during compilation:

```
%Error:  test.v:165:45: Unsupported: Shifting of by over 32-bit number isn't supported. (This isn't a shift of 32 bits, but a shift of 2^32, or 4 billion!)
  165 |     assign out36 = (10'd820 <<< (wire_0 >>> {8'o304, 13'b1001101100, {3{4'o012}}}));
      |                                             ^
        ... See the manual at https://verilator.org/verilator_doc.html?v=5.037 for more assistance.
%Error: Exiting due to 1 error(s)
```

However, in this issue, no such error message was reported—even though a similar large shift amount is involved. Perhaps that's where the problem lies?

sdjasj, 2025-05-10:
It can still be reproduced after [https://github.com/verilator/verilator/pull/5998](https://github.com/verilator/verilator/pull/5998), using Verilator version `Verilator 5.037 devel rev v5.036-50-g100e3d770 (mod)`.

wsnyder, 2025-05-10:
Not surprising as this issue is still open. ;)  Perhaps you meant to try #5993?

closed 2025-05-10 by wsnyder