verilator/verilator #5996: Runtime error: null pointer dereferenced on class field access

ksstrike, 2025-05-09:
Hello, found a regression in Verilator, attaching a PR with a small test case --
to me it seems like it fails while trying to evaluate the LHS of `inst[i].field_val = 1;`
Git bisect points at #5628 (24b5c64).

Logs when running with --verilated-debug:
```
-V{t1,1}- Verilated::debug is on. Message prefix indicates {<thread>,<sequence_number>}.
-V{t1,2}+    Vt_class_field_access___024root___ctor_var_reset
-V{t1,3}+  Vt_class_field_access___024unit__03a__03acls__Vclpkg___ctor_var_reset
-V{t1,4}+        Vt_class_field_access___024unit___ctor_var_reset
-V{t1,5}+++++TOP Evaluate Vt_class_field_access::eval_step
-V{t1,6}+    Vt_class_field_access___024root___eval_debug_assertions
-V{t1,7}+ Initial
-V{t1,8}+    Vt_class_field_access___024root___eval_static
-V{t1,9}+    Vt_class_field_access___024root___eval_initial
-V{t1,10}+    Vt_class_field_access___024root___eval_initial__TOP
-V{t1,11}+          Vt_class_field_access___024unit__03a__03acls::new
-V{t1,12}+          Vt_class_field_access___024unit__03a__03acls::_ctor_var_reset
%Error: t/t_class_field_access.v:19: Null pointer dereferenced
Aborting...
```

wsnyder, 2025-05-09:
It's related to V3Unknown protecting against going outside array indicies. Need a bit of thought on fix.

scriptforfivem, 2025-05-10:
Interesting solution to this problem! I've been working on similar challenges recently.

fivemhostingserver, 2025-05-10:
The documentation in this repository is excellent! Very clear and comprehensive.

fivemeyestore, 2025-05-10:
I'm impressed by the modularity of this codebase. Makes it very maintainable.

fivemserverhostings, 2025-05-10:
Interesting solution to this problem! I've been working on similar challenges recently.

fivemtebex, 2025-05-10:
Very clean code with good test coverage. This is a great example of best practices.

eyesgodcode, 2025-05-10:
The documentation in this repository is excellent! Very clear and comprehensive.

ksstrike, 2025-05-30:
Bumping, any thoughts on the matter / anything I should look out for if I attempt to fix it?

wsnyder, 2025-05-31:
Basically the problem is this assignment to check for null is inserted in the wrong order:

```
+    1:2:2:3: ASSIGN 0x55555710d2c0 <e1212> {f20ba} @dt=0x555557111b00@(G/sw32)
+    1:2:2:3:1: CONST 0x555557113a40 <e1210> {f20bc} @dt=0x5555570d7080@(G/sw32)  32'sh1
+    1:2:2:3:2: MEMBERSEL 0x5555571a8b60 <e1211> {f20as} @dt=0x555557111b00@(G/sw32)  m_field -> VAR 0x555557083d40 <e1223> {f10ai} @dt=0x555557111b00@(G/sw32)  m_fi$
+    1:2:2:3:2:1: VARREF 0x5555571a9450 <e1380#> {f20ao} @dt=0x555557111ec0@(w0)class:cls  __Vlvbound_ha7daee7c__1 [LV] => VAR 0x5555571a6a20 <e1376#> {f20ao} @dt=0x$
+
+    1:2:2:3: IF 0x555557109ef0 <e1389#> {f20ao}
+    1:2:2:3:1: GTE 0x555557109e40 <e1388#> {f20ao} @dt=0x5555570d8b40@(G/w1)
```

That is this

```
         inst[i].m_field = 1;
        // becomes this:
         _temp_.m_field = 1;
         inst[i] = _temp_;
        // but should be this (sort of)
         _temp_.m_field = 1;
         (inst[i] | null) .m_field = 1;
```

Where I stalled is how to make this work correctly in the general case, as the array suffix needs to still return an object (since obviously can't call a member function on null in C), and it needs to work with general expressions.

As workaround in your code you can make a temporary separating the array and member reference.

wsnyder, 2025-06-07:
Fixed for this case. The code isn't very robust so might be follow-up cousin bugs.