verilator/verilator #5999: Declaring a randc non-`packed` struct produces internal error: Unexpected randc variable dtype

sethp, 2025-05-10:
Given `ex.v`:

```verilog
class ex;
    randc struct {
        logic x;
    } s;
endclass: ex

module foo;
  initial begin
    ex e;
    e.randomize(); // this call must be present to trigger the bug
  end
endmodule
```

running `verilator --lint-only ex.v ` produces:

```
%Error: Internal Error: ex.v:4:7: ../V3Randomize.cpp:1463: Unexpected randc variable dtype
                                : ... note: In instance 'foo'
    4 |     } s;
      |       ^
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
```

The following change resolves the error:

```diff
@@ -1,5 +1,5 @@
 class ex;
-    randc struct {
+    randc struct packed {
         logic x;
     } s;
 endclass: ex
```

I don't know enough about the details of the IEEE spec to confidently say that "all structs declared randc must be packed," but if that's true perhaps it would make a more actionable error message than "Internal error"? 

> What 'verilator --version' are you using?  Did you try it with the git master version?

```
$ verilator --version
Verilator 5.034 2025-02-24 rev v5.034
```

> What OS and distribution are you using?

```
$ head -n 5 /etc/os-release
NAME="Manjaro Linux"
PRETTY_NAME="Manjaro Linux"
ID=manjaro
ID_LIKE=arch
BUILD_ID=rolling
```

> May we assist you in trying to fix this in Verilator yourself?

If the fix ought to just be a more-specific error in this case, then yes. I'm not sure when I'll be able to get to it though, so someone else is also certainly welcome to take over if they feel more urgency.

wsnyder, 2025-05-10:
@YilouWang, comments?

YilouWang, 2025-05-12:
Hello, @sethp thanks for mentioning this.

According to IEEE 1800-2023, Section 18.4:

> “An unpacked structure can be declared rand, in which case all of that structure’s random members are solved concurrently using one of the rules listed in this subclause. Unpacked structures shall not be declared randc. A member of an unpacked structure can be made random by having a rand or randc modifier in the declaration of its type.”

So your guess was correct — **randc cannot be applied to unpacked structs**.
As for your suggestion to make the error message more specific instead of throwing a generic error, I think that's a good idea.

@wsnyder, what do you think?