verilator/verilator #6000: Incorrect evaluation of signed comparisons involving ternary expressions and negation

sdjasj, 2025-05-10:
Here is the minimized code `test.v` that triggers the issue:

```verilog
`timescale 1ns/1ps
`define stop   $stop
`define checkd(gotv, expv) \
  do if ((gotv) !== (expv)) begin \
       $write("%%Error: %s:%0d:  got=%0d exp=%0d\n", `__FILE__, `__LINE__, (gotv), (expv)); \
       `stop; \
     end while(0);

module top ();

    reg  in4;
    wire signed wire_2;
    wire out88;

    assign wire_2 = in4 ? 2'b10 : 0;
    assign out88 = (-(wire_2) <= wire_2 ? 1 : 0);

    initial begin
        in4 = 1'b0;
        #5;
        `checkd(out88, '1);
        $write("*-* All Finished *-*\n");
        $finish;
    end

endmodule
```

I ran the following commands:

```
verilator --binary -Wno-lint --timing test.v
./obj_dir/Vtest
```

Verilator produced the output `0`:

```
%Error: test.v:21:  got=0 exp=1
%Error: test.v:21: Verilog $stop
Aborting...
Aborted (core dumped)
```

However, after manual analysis (`in4 = 0` leads to `wire_2 = 0`, so `out88 = 1`) and comparing with Icarus Verilog, the expected result should be `1`.

wsnyder, 2025-05-11:
@gezalore if you could please take a look, passes with `-fno-dfg`. Note 1'sb1 = -1.

gezalore, 2025-05-14:
Nice find, thanks!

PR #6004 (2025-05-14): Fix folding of LteS in DfgPeephole
Fixes #6000