verilator/verilator #6007: Incorrect Optimization Causes Mismatched Output in Verilator

sdjasj, 2025-05-16:
The following is the problematic code `test.v`. I’ve tried my best to minimize it, but it seems the error only occurs during optimization of relatively complex expressions, so it still appears a bit long.

```verilog
`timescale 1ns/1ps
`define stop   $stop
`define checkd(gotv, expv) \
  do if ((gotv) !== (expv)) begin \
       $write("%%Error: %s:%0d:  got=%0d exp=%0d\n", `__FILE__, `__LINE__, (gotv), (expv)); \
       `stop; \
     end while(0);
module top (out20);

    reg  in0;
    reg signed [7:0] in4;
    wire  [1:0] wire_0;
    output wire out20;

    assign wire_0 = in4[0:0] ? ({{7{in4[3:1]}}, 12'd201} & 2'h2) : (!(in0) >> 9'b1111);
    assign out20 = wire_0[0:0];
    initial begin
      #10;
      in4 = 16'b1111_1110;
      in0 = 7'b0;
      #10;
      `checkd(out20, '0);
      $write("*-* All Finished *-*\n");
      $finish; 
    end
endmodule
```

You can reproduce the issue with the following commands:

```
verilator --binary -Wno-lint --timing  test.v
./obj_dir/Vtest
```

Verilator output:

```
%Error: test.v:22:  got=1 exp=0
%Error: test.v:22: Verilog $stop
Aborting...
Aborted (core dumped)
```

However, Icarus Verilog completes successfully with `*-* All Finished *-*`. The value of `out20` is indeed `0`, because `in4 = 16'b1111_1110`, so `in4[0:0] = 0`, which causes `wire_0 = (!(in0) >> 9'b1111)`, resulting in `wire_0 = 0`, hence `out20 = 0`.

During local debugging, I found that enabling the following options makes Verilator produce the correct output:

```
-fno-dfg-peephole
-fno-dfg
-fno-dfg-pre-inline
-fno-expand
-fno-const-bit-op-tree
-fno-dfg-post-inline
```

### System details:
* Verilator version: Verilator 5.037 devel rev v5.036-64-g2f6ecd185
* OS: Ubuntu 22.04.5 LTS (Jammy), Kernel/Release: 22.04

yTakatsukasa, 2025-05-16:
It looks 
https://github.com/verilator/verilator/blob/6b42d789af238eea3b91fd0c69419117f3f4764c/src/V3Number.cpp#L1880
should be 
```c++
    for (uint64_t bit = 0; bit < width(); ++bit) {
```

wsnyder, 2025-05-16:
@sdjasj mind attempting a second pull with @yTakatsukasa (thanks) suggestion please?

sdjasj, 2025-05-16:
No problem, I'm happy to help :) and thank you @yTakatsukasa for your fix suggestion.

yTakatsukasa, 2025-05-16:
https://github.com/verilator/verilator/blob/6b42d789af238eea3b91fd0c69419117f3f4764c/src/V3Number.cpp#L1881
Ah I also think static_cast in this line is not necessary if `bit` is `uint64_t`.

I'm not sure whether an obvious  shortcut of `if (rhsval >= width()) return *this;` is beneficial or not.