verilator/verilator #6011: Issue about function VL_CONCAT_WWW and _vl_insert_WW

godblesszhouzhou, 2025-05-16:
Thanks for taking the time to report this.

Can you attach an example that shows the issue?  (Must be openly licensed, completely self-contained and self-checking so can directly run what you provide. Ideally use test_regress format, see https://veripool.org/guide/latest/contributing.html?highlight=test_regress#reporting-bugs

What 'verilator' command line do we use to run your example?

What 'verilator --version' are you using?  Did you try it with the git master version?

What OS and distribution are you using?

May we assist you in trying to fix this in Verilator yourself?

(Please avoid attaching screenshots that show text - you can convert images to text using e.g. https://ocr.space)

wsnyder, 2025-05-16:
Think you forgot to update the description.

wsnyder, 2025-05-17:
Is "words" properly in-bounds for the variables passed where the function is used?  (Bug even if not).

Does "--runtime-debug" flag any issues around this?

Does "gcc" versus "clang" do differently?  (You mention clang only).  Could well be a clang bug if above two appear clean.

wsnyder, 2025-05-17:
Also if first two are clean, and gcc works, try a newer clang++ version (or a different OS distro with a newer clang++ version).  Which are you using?

wsnyder, 2025-05-19:
You are hitting a new gcc bug.  What version of gcc are you using?  Did you try the latest gcc?

Run gcc with "-Q" on that file only and see where the time is.  You can then post the C file (if sharable) as described here https://gcc.gnu.org/bugs/ along with that profile and they can most likely give you a compiler flag to work around the issue.

We have much larger designs compiling fine so it's not simply a .h size issue; it might be that the gcc people suggest a way Verilator can hack around the speed problem, but it's still their bug fundamentally.

wsnyder, 2025-07-03:
Closing; assuming you've reported this to the gcc maintainers.

closed 2025-07-03 by wsnyder