verilator/verilator #6014: error: member reference base type 'CData' (aka 'unsigned char') is not a structure or union

paul-demo, 2025-05-16:
My example is too large to attach at the moment, but it seems that Verilator is incorrectly pruning out some nested unpacked structs that get generated by PeakRDL. This works in many other designs that use the same type of source code, but I've increased the size of this one today and now the generated Verilator code seems to be making an invalid access into those structs:

```{bash}
Vtop_tcam_ram_hybrid__pi17__DepSet_ha4559ed8__0.cpp:86:76: error: member reference base type 'CData' (aka 'unsigned char') is not a structure or union
   85 |                                                          | (QData)((IData)(vlSymsp->TOP__df_macro_tb__df_macro_i__genblk1__BRA__0__KET____DOT__df_unit_wrap_i__df_unit_i__df_management_ram_i__df_unit_csr_i.__PVT__hwif_out
      |                                                                            ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   86 |                                                                            .__PVT__tcam_ctrl
      |                                                                            ^~~~~~~~~~~~~~~~~

```

I've tried `-fno-inline` on the theory that some names are colliding somewhere in the hierarchy but it also results in the same issue.

paul-demo, 2025-05-16:
Some preceding error messages suggest it's related to structures in the generated CSR code?

```{bash}
Vtop_df_management_ram__pi47__DepSet_hbf71eab4__0.cpp:25:22: error: no viable overloaded '='
   25 |     vlSelf->hwif_out = vlSymsp->TOP__df_macro_tb__df_macro_i__genblk1__BRA__0__KET____DOT__df_unit_wrap_i__df_unit_i__df_management_ram_i__df_unit_csr_i.hwif_out;
      |     ~~~~~~~~~~~~~~~~ ^ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/pswirhun/workplace/CR-195748386/src/Kuiper-DramFifo/build/kuiper_dram_fifo_df_macro_1.5.0/test_txp-cocolator/build/Vtop_df_unit_csr_pkg.h:728:8: note: candidate function (the implicit copy assignment operator) not viable: no known conversion from 'CData' (aka 'unsigned char') to 'const Vtop_df_unit_csr___05Fout_t__struct__0' for 1st argument
  728 | struct Vtop_df_unit_csr___05Fout_t__struct__0 {
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/pswirhun/workplace/CR-195748386/src/Kuiper-DramFifo/build/kuiper_dram_fifo_df_macro_1.5.0/test_txp-cocolator/build/Vtop_df_unit_csr_pkg.h:728:8: note: candidate function (the implicit move assignment operator) not viable: no known conversion from 'CData' (aka 'unsigned char') to 'Vtop_df_unit_csr___05Fout_t__struct__0' for 1st argument
  728 | struct Vtop_df_unit_csr___05Fout_t__struct__0 {
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Vtop_df_management_ram__pi47__DepSet_hbf71eab4__0.cpp:539:19: error: assigning to 'CData' (aka 'unsigned char') from incompatible type 'Vtop_df_unit_csr___05Fin_t__struct__0'
  539 |         = vlSelf->hwif_in;
      |           ~~~~~~~~^~~~~~~
```

paul-demo, 2025-05-16:
Some earlier error messages suggest it's struggling with the unpacked structures in the generated CSR. This design has an array of 4 blocks which each have an identical generated CSR block, but that has not been a problem for any version of Verilator I've used in the preceding ~12 months. Using Verilator 5.024 2024-04-05 rev UNKNOWN.REV now.

```{bash}
Vtop_df_management_ram__pi47__DepSet_hbf71eab4__0.cpp:25:22: error: no viable overloaded '='
   25 |     vlSelf->hwif_out = vlSymsp->TOP__df_macro_tb__df_macro_i__genblk1__BRA__0__KET____DOT__df_unit_wrap_i__df_unit_i__df_management_ram_i__df_unit_csr_i.hwif_out;
      |     ~~~~~~~~~~~~~~~~ ^ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
{path}/build/Vtop_df_unit_csr_pkg.h:728:8: note: candidate function (the implicit copy assignment operator) not viable: no known conversion from 'CData' (aka 'unsigned char') to 'const Vtop_df_unit_csr___05Fout_t__struct__0' for 1st argument
  728 | struct Vtop_df_unit_csr___05Fout_t__struct__0 {
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
{path}/build/Vtop_df_unit_csr_pkg.h:728:8: note: candidate function (the implicit move assignment operator) not viable: no known conversion from 'CData' (aka 'unsigned char') to 'Vtop_df_unit_csr___05Fout_t__struct__0' for 1st argument
  728 | struct Vtop_df_unit_csr___05Fout_t__struct__0 {
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Vtop_df_management_ram__pi47__DepSet_hbf71eab4__0.cpp:539:19: error: assigning to 'CData' (aka 'unsigned char') from incompatible type 'Vtop_df_unit_csr___05Fin_t__struct__0'
  539 |         = vlSelf->hwif_in;
      |           ~~~~~~~~^~~~~~~
```

wsnyder, 2025-05-16:
We'll almost certainly need aan openly licensed, completely self-contained test case so we can directly run what you provide.  Ideally use test_regress format, see [Reporting Verilator Bugs](https://veripool.org/guide/latest/contributing.html?highlight=test_regress#reporting-bugs) and especially section on use of sv-bugpoint which can reduce your large example.

paul-demo, 2025-05-16:
Yes, I'm not sure if or how the issue would be reproducible in an excerpt of this design. Perhaps surprisingly VScode also picks up on the issue and puts squiggle error lines under the assignment here:

[image]

This shows that it thinks this structure `hwif_out` is of type CData when it is supposed to be some fairly large nested unpacked struct generated by PeakRDL.
```
vlSymsp->TOP__df_macro_tb__df_macro_i__genblk1__BRA__0__KET____DOT__df_unit_wrap_i__df_unit_i__df_management_ram_i__df_unit_csr_i.hwif_out
```

paul-demo, 2025-05-16:
Anyway, pretty conclusive here in the generated .h file:
```{c++}
    VL_IN8(hwif_in, 0, 0);
    VL_OUT8(hwif_out, 0, 0);
```

Those are actually large unpacked structs. I'm wondering if it's related to the order in which the .sv and _pkg.sv file are listed on the command line or something like that, and it's just choosing a CData type as a placeholder instead of the real struct that's defined in the _pkg.sv file.

paul-demo, 2025-05-17:
Well for whatever reason, upgrading from Verilator 5.024 -> 5.036 fixed it and those are now the expected type that they had been all along in 5.024 prior to today. I'll just close this since I don't want to worry about why v5.024 didn't work.
```{c++}
    Vtop_df_unit_csr___05Fin_t__struct__0 hwif_in;
    Vtop_df_unit_csr___05Fout_t__struct__0 hwif_out;
```

closed 2025-05-17 by paul-demo