verilator/verilator #6018: Fix scope randomized failure

wsnyder, 2025-05-17:
@toddstrader, from #5958 (presumably) the RTLmeter regression here https://github.com/verilator/verilator/actions/runs/15081043632/job/42397728561 failed running OpenPiton and others.  Can you please debug?  Thanks.

```
Vsim___024root__DepSet_hb1836b75__5.cpp:3219:83: error: use of undeclared identifier '__VscopeHash'
   16.47 |  3219 |                 vlSelf->__Vfunc_swendian64__131__out = VL_SCOPED_RAND_RESET_Q(64, __VscopeHash, 10517932497710372378ull);
```

toddstrader, 2025-05-17:
Yeah, I'll check it out.

wsnyder, 2025-05-17:
Also, while in there, is there it seems there's some instability based on pointer names in the Xrand names generated, as I changed the code but the same AST into the algorithm is getting a different name.

```
======================================================================
= t_x_rand_stability.old/Vt_x_rand_stability_036_tristate.tree <-> t_x_rand_stability.new/Vt_x_rand_stability_036_tristate.tree
======================================================================
= t_x_rand_stability.old/Vt_x_rand_stability_037_unknown.tree <-> t_x_rand_stability.new/Vt_x_rand_stability_037_unknown.tree
--- /tmp/4086946_t_x_rand_stability.old_Vt_x_rand_stability_037_unknown.tree.a  2025-05-17 16:45:24.065300686 -0400
+++ /tmp/4086946_t_x_rand_stability.new_Vt_x_rand_stability_037_unknown.tree.b  2025-05-17 16:45:24.065300686 -0400
@@ -6,15 +6,15 @@
     1:2:1:1: VARREF 0x <e> {f13al} @dt=0x@(G/w1)  clk [RV] <- VAR 0x <e> {f13al} @dt=0x@(G/w1)  clk [PI] INPUT [P] [VSTATIC]  PORT
     1:2: VAR 0x <e> {f13al} @dt=0x@(G/w1)  clk [PI] INPUT [P] [VSTATIC]  PORT
     1: MODULE 0x <e> {f8ai}  t  L2 [1ps]
-    1:2: VAR 0x <e> {f41ax} @dt=0x@(G/w32)  __Vxrand_h87950a6c__0 [VSTATIC]  XTEMP
+    1:2: VAR 0x <e> {f41ax} @dt=0x@(G/w32)  __Vxrand_h8794da3b__0 [VSTATIC]  XTEMP
     1:2: INITIAL 0x <e> {f41ax}
     1:2:2: ASSIGN 0x <e> {f41ax} @dt=0x@(G/w32)
     1:2:2:1: OR 0x <e> {f41ax} @dt=0x@(G/w32)
     1:2:2:1:1: CONST 0x <e> {f41ax} @dt=0x@(G/w32)  32'h0
     1:2:2:1:2: AND 0x <e> {f41ax} @dt=0x@(G/w32)
     1:2:2:1:2:1: CONST 0x <e> {f41ax} @dt=0x@(G/w32)  32'hffffffff
-    1:2:2:1:2:2: VARREF 0x <e> {f41ax} @dt=0x@(G/w32)  __Vxrand_h87950a6c__0 [RV] <- VAR 0x <e> {f41ax} @dt=0x@(G/w32)  __Vxrand_h87950a6c__0 [VSTATIC]  XTEMP
-    1:2:2:2: VARREF 0x <e> {f41ax} @dt=0x@(G/w32)  __Vxrand_h87950a6c__0 [LV] => VAR 0x <e> {f41ax} @dt=0x@(G/w32)  __Vxrand_h87950a6c__0 [VSTATIC]  XTEMP
+    1:2:2:1:2:2: VARREF 0x <e> {f41ax} @dt=0x@(G/w32)  __Vxrand_h8794da3b__0 [RV] <- VAR 0x <e> {f41ax} @dt=0x@(G/w32)  __Vxrand_h8794da3b__0 [VSTATIC]  XTEMP
+    1:2:2:2: VARREF 0x <e> {f41ax} @dt=0x@(G/w32)  __Vxrand_h8794da3b__0 [LV] => VAR 0x <e> {f41ax} @dt=0x@(G/w32)  __Vxrand_h8794da3b__0 [VSTATIC]  XTEMP
     1:2: VAR 0x <e> {f13al} @dt=0x@(G/w1)  clk INPUT [VSTATIC]  PORT
     1:2: VAR 0x <e> {f14aj} @dt=0x@(G/sw32)  cyc [VSTATIC]  VAR
     1:2: INITIALSTATIC 0x <e> {f14ap}
@@ -60,7 +60,7 @@
     1:2:1:1: SENITEM 0x <e> {f40ao} [POS]
     1:2:1:1:1: VARREF 0x <e> {f40aw} @dt=0x@(G/w1)  clk [RV] <- VAR 0x <e> {f13al} @dt=0x@(G/w1)  clk INPUT [VSTATIC]  PORT
     1:2:2: ASSIGNDLY 0x <e> {f41au} @dt=0x@(G/w32)
-    1:2:2:1: VARREF 0x <e> {f41ax} @dt=0x@(G/w32)  __Vxrand_h87950a6c__0 [RV] <- VAR 0x <e> {f41ax} @dt=0x@(G/w32)  __Vxrand_h87950a6c__0 [VSTATIC]  XTEMP
+    1:2:2:1: VARREF 0x <e> {f41ax} @dt=0x@(G/w32)  __Vxrand_h8794da3b__0 [RV] <- VAR 0x <e> {f41ax} @dt=0x@(G/w32)  __Vxrand_h8794da3b__0 [VSTATIC]  XTEMP
     1:2:2:2: VARREF 0x <e> {f41aj} @dt=0x@(G/w32)  x_assigned [LV] => VAR 0x <e> {f17as} @dt=0x@(G/w32)  x_assigned [VSTATIC]  VAR
     1:2:2: ASSIGNDLY 0x <e> {f45an} @dt=0x@(G/sw32)
     1:2:2:1: ADD 0x <e> {f45au} @dt=0x@(G/sw32)
```

Which causes a test failure
-  

```
diff --git a/test_regress/t/t_x_rand_stability.out b/test_regress/t/t_x_rand_stability.out
index 562cb9864..5a1ac278d 100644
--- a/test_regress/t/t_x_rand_stability.out
+++ b/test_regress/t/t_x_rand_stability.out
@@ -12,6 +12,6 @@ rand = 0xe85acf2d
 rand = 0x15e12c6a
 rand = 0x0f7f28c0
 rand = 0xe189c52a
-x_assigned = 0xc0391efd
+x_assigned = 0xf9f54cf5
 Last rand = 0xf0700dbf
 *-* All Finished *-*

```

wsnyder, 2025-05-18:
I wasn't able to find a seed that was stable, and with the above large project problems figured I'd relieve you of the stress of needing a quick fix.  So I backed this out, please add back the files in [640339e](https://github.com/verilator/verilator/commit/640339ec363d2df7dfd3fdf629924dc649509b4f) to revert-the-revert in your pull.

toddstrader, 2025-05-19:
Are there notes somewhere on RTLMeter?  More specifically, is there a simple way to run this locally?  It doesn't look like the Makefiles know about it:
```
$ git grep -il rtlmeter
.github/workflows/reusable-rtlmeter-build.yml
.github/workflows/reusable-rtlmeter-run.yml
.github/workflows/rtlmeter.yml
```
cc @gezalore

toddstrader, 2025-05-19:
Oh, just found:
https://github.com/verilator/rtlmeter
Reading . . .

gezalore, 2025-05-19:
In increasing order of specificity:
https://github.com/verilator/rtlmeter
https://verilator.github.io/rtlmeter/
https://verilator.github.io/rtlmeter/running.html
https://verilator.github.io/rtlmeter/running.html#rerunning-external-commands-manually

TLDR: clone https://github.com/verilator/rtlmeter, put verilator on PATH then `./rtlmeter run --cases OpenPiton:1x1:hello`. If needed, use the printed command in the printed directory to rerun manually.

Let me know if something is obtuse in the docs, I'm happy to improve it.