verilator/verilator #6022: How to Generate a Shared Library (.so) from RTL Code Using Verilator

alpha-beta-user, 2025-05-19:
Problem Description
I'm working on converting Verilog/SystemVerilog RTL code to C++ models using Verilator and then compiling these models into shared libraries (.so files). My goal is to integrate these shared libraries into another C/C++ application for use in a larger simulation environment.

What I've Tried
I've successfully converted RTL code to C++ using Verilator with:
However, I'm uncertain about the correct approach to compile the generated C++ files into a shared library. 
Specifically:
What Verilator parameters should be used to specify shared library generation?
Is it necessary to write wrapper code to export the proper interfaces?
Do I need to create specific header files for external applications?

Environment
Verilator version: 5.36
Operating system: ubutun 20
Compiler: GCC  8

Specific Questions
Are there direct command-line options in Verilator to generate .so files?
--lib_create not work. it report error about == operator for sc_core::sc_signal

If manual compilation is needed, what is the complete compilation workflow?
How can I properly export symbols from the shared library for external program calls?
Are there any examples of using the generated shared library in C/C++ applications?
Thank you for any guidance or examples!

wsnyder, 2025-05-19:
I don't have any recopies for generating shared objects, you'll need to make a custom Makefile (or cmake) to do that.

The error you get is likely because you'll also need to build a separate shared object for the SystemC library.

Note that Verilator is not designed to have multiple models of different versions present at the same time, so if you're trying to make a shared object that can be present with other Verilator shared objects, beware of that.  (And even more so SystemC can't have multiple versions present.)

Closing but feel free to follow up. If you want to post your results here it might help the next person.

closed 2025-05-19 by wsnyder

alpha-beta-user, 2025-05-20:
Thank you!

Yes, I have solved. I will later share my script and method. Hoping it can help next persion