verilator/verilator #6023: Bit width truncation due to signed int return type causes incorrect signal width interpretation

sdjasj, 2025-05-19:
Consider the following code `test.v`, which contains a signal with an extremely large bit-width (exceeding the range representable by a 32-bit `int`):

```verilog
`timescale 1ns/1ps

module top (out0);
  output wire [32'hFFFF_FFFF:1] out0;

  assign out0 = 1;

  initial begin
    #10;
    $display("out0: %b", out0);
    $finish; 
  end

endmodule
```

When running the following commands:

```sh
verilator --binary -Wno-lint --timing test.v
./obj_dir/Vtest
```

The code compiles successfully (the excessively large bit-width seems to bypass Verilator’s width checks, such as `Width of bit range is huge; vector of over 1 billion bits:` or `Value too wide for 32-bits expected in this context 36'haffffffff`), and treats `out0` as a 3-bit signal. The output is:

```
out0: 001
```

So it appears that somewhere in the code, a conversion from `uint32_t` to `int` occurs, which causes `32'hffff_ffff` to be interpreted as `-1`.

I further looked into the source code and found the following problematic line:

https://github.com/verilator/verilator/blob/master/src/V3Width.cpp#L949

Here, the `width` value should ideally be of type `uint32_t`.

After further investigation, it turns out that the cause is the following functions, which return `int` instead:

https://github.com/verilator/verilator/blob/master/src/V3AstNodeOther.h#L2698-L2710

A straightforward fix would be to change the return type of these functions to `uint32_t`. However, I’m not sure how extensive the impact would be on the rest of the codebase. If it's safe and has no side effects, I’d be happy to submit a PR to fix this.

wsnyder, 2025-05-19:
int vs uint32_t (vs size_t in C++) inconsistencies are rampant, and not really worth cleaning up as this won't happen with real code.  Might you make a pull to warn over 1<<31 instead?

wsnyder, 2025-05-19:
BTW part of the problem will always be that IEEE says part select indices are signed 32-bits, while C++ is size_t - 64 bit unsigned.

sdjasj, 2025-05-19:
> int vs uint32_t (vs size_t in C++) inconsistencies are rampant, and not really worth cleaning up as this won't happen with real code. Might you make a pull to warn over 1<<31 instead?

Okay, let me see how to warn appropriately

closed 2026-10-21 by wsnyder