verilator/verilator #6029: Constant values zero in sim while they should be non-zero unsigned numbers

BertVerrycken, 2025-05-21:
## Issue

### Example

The issue cannot be reproduced with a simple example (I tried).

### Description

Package with definitions:
```
  const word15_u_t C_NAN_BF16     = 15'h7FC0;
  const word15_u_t C_INF_BF16     = 15'h7F80;
  const word15_u_t C_ONE_BF16     = 15'h3F80;
```

Design using the package constants:
```
  const word31_u_t          c_nan  = {C_NAN_BF16, 16'h0};
  const word31_u_t          c_zero = 0;
  const word31_u_t          c_one  = {C_ONE_BF16, 16'h0};
```

In a contained example, the constants have the expected value. In my design (which I cannot disclose), the constants c_one and c_nan are both all zeroes. The vcd wavedump shows the wrong values and the output is wrong. Questa OTH simulates the exact same test with exact the same code correct. Honestly, I have no idea why these constants get set to zero instead of the correct values, any pointers to how this can be recreated is appreciated. Side note is that it used to work before when the code was for 16-bit wide numbers, eg:

```
  const word15_u_t          c_nan  = C_NAN_BF16;
  const word15_u_t          c_zero = 0;
  const word15_u_t          c_one  = C_ONE_BF16;
```

## Verilator

### Command

This sim is run with cocotb:
```
/usr/local/bin/verilator -cc --exe -Mdir sim_build -DCOCOTB_SIM=1 --top-module tb --timing -Wno-UNOPTFLAT -Wno-ASSIGNIN --timescale 1ns/1ps --vpi --public-flat-rw --prefix Vtop -o Vtop -LDFLAGS "-Wl,-rpath,lib/python3.12/site-packages/cocotb/libs -Llib/python3.12/site-packages/cocotb/libs -lcocotbvpi_verilator"  ../../float/vlog/float_pkg.sv ../../tanh_bf16/vlog/tanh_bf16.sv ../vlog/tanh_bf16_array.sv tb.sv lib/python3.12/site-packages/cocotb/share/lib/verilator/verilator.cpp
```

### Version

```
Verilator 5.036 2025-04-27 rev v5.036
```

## OS and distribution

```
NAME="Rocky Linux"
VERSION="9.5 (Blue Onyx)"
```

## Assistance

I'm afraid my background is really systemverilog and not software, I doubt I will be able to change any code myself. However, I will support any debugging effort from my side if feasible.

wsnyder, 2025-05-21:
Are they used in modules with parameters of two different flavors? If so could be #4920.

To do more we'll need a completely self-contained test case so we can directly run what you provide.  Ideally use test_regress format, see [Reporting Verilator Bugs](https://veripool.org/guide/latest/contributing.html?highlight=test_regress#reporting-bugs) and especially section on use of sv-bugpoint, you can have it look for wrong value (change waves to a $display and look for that).  `fno-inline` might help it reproduce smaller.

BertVerrycken, 2025-05-21:
They are const definitions not parameters but I looked at the other buf report and that is something interesting.

The top is an array of another module, where the tanh_bf16 module that is instantiated 32 times has its parameters as default in the instantiation:

```
module tanh_bf16_array
  import float_pkg::*;
#(int                           NOM         = 32  // Number of multipliers
)
(
  // Operands and result
  input  word16_u_t [NOM-1:0]   operand,          // Input  operand
  output word16_u_t [NOM-1:0]   result            // Output result
);

  generate
    for (genvar i = 0; i < NOM; i++) begin: gen_tanh_bf16_array
      tanh_bf16 i_tanh_bf16 (
        .operand    (operand[i]),
        .result     (result[i])
      );
    end
  endgenerate

endmodule
```

The default is a parameter and a local parameter both of different types (logic and int):
```
module tanh_bf16
  import float_pkg::*;
#(parameter logic FLOAT_ROUNDING  = 1,                        // Enable float rounding
  localparam int  RDW             = FLOAT_ROUNDING ? 16 : 32  // Result data width: BF16 = 16 bits with 7 guard bots
)
(
  // BF16 inputs and output
  input  word16_u_t               operand,                    // Input  operand
  output logic unsigned [RDW-1:0] result                      // Output result
);
```

The parameter changes the width of the output result unsigned.

Will look into sv-bugpoint and get back to you.

wsnyder, 2025-07-03:
Please reopen if provide a small testcase.

closed 2025-07-03 by wsnyder