verilator/verilator #6031: /V3EmitCFunc.h:1457: Unknown node type reached emitter: CVTPACKEDTOARRAY

paul-demo, 2025-05-22:
Verilator 5.036 2025-04-27 rev UNKNOWN.REV

Unexpected crash from packed array assignment involving byte enddianness swap. This works in 5.024 but not in 5.036. I tried writing a test_regress example but don't have the time to figure out how to get that to work. A version of the affected code with variables names changed is below:

```{verilog}
package some_pkg;

  typedef struct packed {
    logic [15:0]  a;
    logic [2:0]   b;
    logic [4:0]   c;
    logic [20:0]  d;
    logic [2:0]   e;
    logic [15:0]  f;
    logic         g;
    phaser_type_t h;
    logic         i;
    logic         j;
  } some_t;

  function logic [7:0][7:0] some_func(input some_t s);
    logic [7:0][7:0] x;
    begin
      x[0][0]   = s.j;
      x[0][1]   = s.i;
      x[0][6:2] = s.h;
      x[0][7]   = s.g;
      x[2:1]    = {<<8{s.f}};
      x[3][7:5] = s.e;
      x[3][4:0] = s.d[20:16];
      x[5:4]    = {<<8{s.d[15:0]}};
      x[7:6]    = {<<8{s.a}};
      return x;
    end
  endfunction

endpackage
```

```{bash}
/V3EmitCFunc.h:1457: Unknown node type reached emitter: CVTPACKEDTOARRAY
  187 |       x[2:1]    = {<<8{s.d}};
      |                    ^~
```

wsnyder, 2025-05-22:
Could you make this self checking though please so when we fix the error we have some confidence in the value?

paul-demo, 2025-05-22:
Yes, I can try again later this week. It's an error at verilation but when I tried to put it in a test-regress target, it just ran forever.

paul-demo, 2025-05-22:
Possibly related issue: https://github.com/verilator/verilator/issues/5955

wsnyder, 2025-05-22:
Oh, #5955 isn't in your version, please run on master.

paul-demo, 2025-05-22:
Yeah, definitely looks to be the same issue -- streaming to/from packed arrays.

For now, I just removed that from the source code:

```{verilog}
x[1] = s.f[15:8];
x[2] = s.f[7:0]
```

closed 2025-05-22 by paul-demo