verilator/verilator #6032: test_regress targets don't return and exit on MacOS when done

paul-demo, 2025-05-22:
This might be user error because I don't often interact with Verilator's unit tests but: on mainline on two different Macs I see that the test-regress targets don't exit when they finish. It just gets stuck in some loop thinking the test is still running. Apologies in advance if I'm just doing something wrong here -- trying to contribute a unit test here.

As you can see here I did a ctrl+c to kill it after ~30 seconds, but it still thinks that the command `t/t_lint_iface_topmodule1.py --debug` is running.

[image]

paul-demo, 2025-05-22:
Possibly a lingering cross-platform issue related to https://github.com/verilator/verilator/issues/5752

wsnyder, 2025-05-22:
Thanks for reporting. Note the primary developers don't use Macs, so a debug and pull to fix would be especially welcome.