verilator/verilator #6045: Struct assignment misinterpreted as concatenation

dpetrisko, 2025-05-28:
> Thanks for taking the time to report this.
> 
> Can you attach an example that shows the issue?  (Must be openly licensed, completely self-contained and self-checking so can directly run what you provide. Ideally use test_regress format, see https://veripool.org/guide/latest/contributing.html?highlight=test_regress#reporting-bugs

This seems like a verilator lint bug, but I wouldn't be surprised if undefined by spec. Happy to convert to a test_regress if this is an actual bug

testcase.sv:
```
module testcase;

    typedef enum logic [1:0] { e_0, e_1, e_2, e_3 } enum_e;

    typedef struct packed {
        integer unsigned x;
        integer unsigned y;
    } foo_s;

    typedef struct packed {
        integer unsigned y;
    } bar_s;

    localparam foo_s foo = '{
        y: (1 << e_0) | (1 << e_3)
        ,default: '0
    };

    localparam bar_s bar = '{
        y: (1 << e_0) | (1 << e_3)
    };

endmodule
```

> What 'verilator' command line do we use to run your example?

```
verilator --lint-only -Wall -Wno-UNUSEDPARAM testcase.sv
...
%Warning-WIDTHCONCAT: testcase.sv:12:23: Unsized numbers/parameters not allowed in concatenations.
                                       : ... note: In instance 'testcase'
   12 |         y: (1 << e_0) | (1 << e_3)
      |                       ^
                      ... For warning description see https://verilator.org/warn/WIDTHCONCAT?v=5.036
                      ... Use "/* verilator lint_off WIDTHCONCAT */" and lint_on around source to disable this message.
%Error: Exiting due to 1 warning(s)
```

> What 'verilator --version' are you using?  Did you try it with the git master version?

Verilator 5.037 devel rev v5.036-110-g5fcd0e52e (current master)

> What OS and distribution are you using?

Centos7

> May we assist you in trying to fix this in Verilator yourself?

If trivial, sure! If non-trivial it will be low priority since it's just a warning.

(Please avoid attaching screenshots that show text - you can convert images to text using e.g. https://ocr.space)

paul-demo, 2025-05-29:
Definitely looks fishy to me that Verilator thinks that is a WIDTHCONCAT error. But, I recently read that according to the LRM or at least Xilinx's representation of it (https://in.ncu.edu.tw/ncume_ee/digilogi/vhdl/Verilog_Reference_Guide.pdf) that in the expressions `i << j` and `i >> j` , the width of `j` is "self-determined" (but, implicitly, `i` is not). So I think you would be relying upon `foo.y` projecting the type of `y` as `integer unsigned` to determine what the width of `1 << e_0` really means.

Not sure if this is a toy example, but there are a lot of work around possibilities too -- for one thing, why have both binary and 1-hot representations of the same thing? Most synthesis tools are able to map to different FSM encodings without any source code changes, regardless of whether you use binary or one-hot encoding in your enum. Another thing to try is just ` y: (4'h1 << e_0) | (4'h1 << e_3)` rather than what you have and see if it still has an unknown-width error in that case.

dpetrisko, 2025-05-29:
>Definitely looks fishy to me that Verilator thinks that is a WIDTHCONCAT error. But, I recently read that according to the LRM or at least Xilinx's representation of it (https://in.ncu.edu.tw/ncume_ee/digilogi/vhdl/Verilog_Reference_Guide.pdf) that in the expressions i << j and i >> j , the width of j is "self-determined" but the width of i is not. So I think you would be relying upon foo.y projecting the type of y as integer unsigned onto 1 in order to determine what the width of 1 in 1 << e_0 really means.

Yeah, I might expect a WIDTHEXPAND warning for the line y: (1 << e_0) | (1 << e_3) as y is 32b and (1 << e_0) may end up as 2b. Either way, it is a known value and not a concatenation operation

>Not sure if this is a toy example or your real use only has 4 enum values

Toy. This pattern is commonly used for packing named feature flags. A more clear example:

```
  enum logic [1:0] { e_support_foo , e_support_bar, e_support_baz, e_support_gaz } features_e;

  typedef struct packed {
        integer unsigned widget0;
        integer unsigned widget1;
   } features_s;

  localparam features_s features = '{
    widget0: (1 << e_support_foo) | (1 << e_support_baz),
    widget1: (1 << e_support_gaz)
};

...

  if (features.widget0[e_support_foo]) begin
    // Do foo stuff
  end

```

wsnyder, 2025-05-31:
The concat is to form the value of the packed struct.  Generally I think any concat made as a result of a packed structure can have this warning disabled. 

I was going to suggest how to make a pull, but when starting to describe it realized it's just a few lines, so fixed!

closed 2025-05-31 by wsnyder