verilator/verilator #6054: lint_off cannot find file waiver regression between 5.024 and 5.036

paul-demo, 2025-05-29:
I found this waiver works for ignoring a missing module using `--lint-only` mode in 5.024 but not for 5.036.

```
`verilator_config
lint_off -file "*.sv" -match "Cannot find file containing module: 'some_module_name'"
```

I'll try to make a test-regress example now to demonstrate.

paul-demo, 2025-05-29:
Please see https://github.com/verilator/verilator/pull/6055

I also made a commit off of v5.024 using the older perl style test-regress script, but didn't submit that as a PR since it seems in the interim things have moved over to python scripts. But, the same .vlt file successfully waives the missing module issue on 5.024.

wsnyder, 2025-05-29:
You aren't supposed to be able to suppress unnamed errors, that was a bug that it worked.

If you really need to specifically suppress this error was can give it a name you can then suppress.

paul-demo, 2025-05-29:
For us to use this as we have been for linting I'd really like to be able to waive module-not-found as, I guess accidentally, it was supported in prior releases. We could alternatively make a bunch of dummy RTL versions of every third party IP that isn't available in Verilator-consumable source code format (every primitive component in an FPGA comes to mind...), but that's a lot more labor and also then there's the risk that someone tries to use the dummy wrapper rather than depend on the real version. A lot of IP uses VHDL also which gets wrapped in verilog wrappers, so we would have to wrap every one of those in some dummy piece of verilog in order to lint designs that include it.

I guess a lot of cases are waiving things like Xilinx XPMs which do have an RTL equivalent, but it's tied up with a particular (licensed) piece of installed tooling; not something we would want to extract and host in regular source code format. Also, if we have to hit a license server to spin up Vivado in order to use Verilator as a lint tool, then most of the benefits of open source are gone and we may as well just hit a license server for Spyglass lint or something commercial like that.

paul-demo, 2025-05-29:
I'll add that the way I think this works with commercial offerings (like between AMD Vivado and Synopsys VCS), is that there's some length build process whereby enctrypted AMD IP gets built into some simulate-able format that VCS can read but users cannot. So there's the possibility for commercial lint tools like those from Synopsys to be able to lint against these IPs. For Verilator, since the most it could do in that case is recognize a block of encrypted content, I'm not sure waiving the whole module is any less coverage than trying to pass the simulation libraries through Verilator.

closed 2025-05-30 by wsnyder