verilator/verilator #6076: FST Trace hierarchicalName.rfind(' ') runtime errors

paul-demo, 2025-06-09:
I commented on https://github.com/verilator/verilator/issues/5668 which is closed but I'm reopening the issue with this new bug report. This issue showed up for me using Verilator 5.036 when it did not show up with older versions like 5.022. Encountered when using VPI/Cocotb, but I think it may be FST/trace related.

I basically did the manual version of sv-bugpoint and repeatedly simplified the example until it contains no extraneous lines of code except the one line which causes the error. See below:

```systemverilog
package foo_pkg;
  function int foo_func;
    input int b;
    // INCLUDING THE FOLLOWING LINE RESULTS IN THE FOLLOWING RUNTIME ERROR:
    // Assertion failed: (hierarchicalName.rfind(' ') != std::string::npos), function declare, file verilated_fst_c.cpp, line 197.
    // EXCLUDING THE DECLARATION OF b_current MAKES THE ERROR GO AWAY!
    /////////////////////
    int b_current;
    /////////////////////

    begin
      return 0;
    end
  endfunction
endpackage

module dut
  import foo_pkg::*;
(
    input  logic i,
    output logic o
);
  assign o = i;
endmodule

module dut_tb (
    input  logic i,
    output logic o
);
  dut u_dut (.*);
endmodule
```

paul-demo, 2025-06-09:
Redacted build flags:
```bash
INFO: Running command in directory /some/path/build:
	perl
		/opt/homebrew/bin/verilator
		-cc
		--exe
		-Mdir
		/some/path/build
		-DCOCOTB_SIM=1
		--top-module
		dut_tb
		--vpi
		--prefix
		Vtop
		-o
		dut_tb
		-LDFLAGS
		'-Wl,-rpath,/user_home/.venv/lib/python3.9/site-packages/cocotb/libs
		-L/user_home/.venv/lib/python3.9/site-packages/cocotb/libs
		-lcocotbvpi_verilator'
		--trace
		--public-params
		-Wno-UNOPTTHREADS
		--public-flat-rw
		--timescale
		1ns/1ps
		--build
		-MAKEFLAGS
		-j4
		-MAKEFLAGS
		'OPT_FAST="-O2"'
		--trace-fst
		--trace-structs
		-I/some/path/src/some_dir
		-CFLAGS
		-Wno-deprecated
		/user_home/.venv/lib/python3.9/site-packages/cocotb/share/lib/verilator/verilator.cpp
		/some/path/src/some_dir/test/dut_tb.sv
```

The C++ build completes successfully:
```bash
- V e r i l a t i o n   R e p o r t: Verilator 5.036 2025-04-27 rev UNKNOWN.REV
- Verilator: Built from 0.026 MB sources in 4 modules, into 0.040 MB in 13 C++ files needing 0.000 MB
- Verilator: Walltime 3.557 s (elab=0.004, cvt=0.011, bld=3.523); cpu 0.015 s on 1 threads
INFO: Running
INFO: [edalize.cocolator] run_main()
INFO: [edalize.cocolator] Running with pytest
```

Then cocotb starts and it fails at runtime due to the Verilator c++ file where the hierarchicalName.rfind(' ') line exists:
```bash
INFO: Running command
    /some/path/build/dut_tb --trace in directory/some/path
     -.--ns INFO     gpi                                ..mbed/gpi_embed.cpp:109  in set_program_name_in_venv        Using Python virtual environment interpreter at /user_home/.venv/bin/python
     -.--ns INFO     gpi                                ../gpi/GpiCommon.cpp:101  in gpi_print_registered_impl       VPI registered
     0.00ns INFO     cocotb                             Running on Verilator version 5.036 2025-04-27
     0.00ns INFO     cocotb                             Running tests with cocotb v1.9.2 from /user_home/.venv/lib/python3.9/site-packages/cocotb
     0.00ns INFO     cocotb                             Seeding Python random module with supplied seed 1400770863
     0.00ns INFO     cocotb.regression                  Found test test.foo_test.my_test
     0.00ns INFO     cocotb.regression                  running my_test (1/1)
Assertion failed: (hierarchicalName.rfind(' ') != std::string::npos), function declare, file verilated_fst_c.cpp, line 197.
FAILED

```

wsnyder, 2025-06-10:
Great, with that test case I see what's going on.  Need to re-sort how the trace elements are declared, will fix this within a few days.

gezalore, 2025-06-10:
Not sure what you mean by "re-sort" the declaration, but keep in mind all every scope should only be entered once (children within a scope can be declared/descended into in arbitrary order). If it helps I can look into this tomorrow, so you don't have to.