verilator/verilator #6089: Using hierarchical verilation to modify specific module, plug back in

jiwoop, 2025-06-14:
Hierarchical verilation allows us to separate certain module out of the overall flat model. Would it be possible to modify that separate module in C++, add features, and plug that back in, and simulate the system with Verilator?

wsnyder, 2025-06-14:
Possible? Certainly, it's just C++ code.  Something we're intending to support? Not currently and unlikely, meaning every release you might need to change how you do the changes.

Alternatively consider having Verilator make a SystemC model, and you can then make other SystemC models.  You'll need to make the top level in SystemC yourself though then.

Closing as answered but feel free to follow up here.

closed 2025-06-14 by wsnyder

jiwoop, 2025-06-15:
Thanks for the response. One case I was skeptical was, what if I were to change I/Os of the module (in generated C++)? e.g., adding input wires or output ports. Then wouldn't that mess with the symbol table implementation?

wsnyder, 2025-06-15:
You can add/remove I/O as you wish if not using the DPI nor VPI. If using those you need to also update the constructor's calls that define the signals for the DPI/VPI.  Never harm to adding (just won't show up in VPI.) You should get a compile error if delete a signal that matters to VPI/DPI, so easy to find.

wsnyder, 2025-06-16:
As above this is not something we intend to support, there is not and will not be a clean interface.  From what you describe it sounds you should consider SystemC instead.

wsnyder, 2025-06-17:
SystemC is nicer in that it is time-aware and sensitive to signals ;)   Also prevents races between IO signals.

JiwooPark-arm, 2025-07-17:
Hello, coming back at this... would DPI interface be helpful for this purpose? Maybe to override ico/nba/act_sequent functions in hierarchically Verilated C++ module?