verilator/verilator #6105: Internal Error: ../V3Ast.h:2656: AstNode is not of expected type, but instead has type 'DOT'

solomatnikov, 2025-06-21:
```
verilator --top-module tb_axi_xbar --binary --timing sv-bugpoint-minimized.sv -Wno-fatal
%Error: sv-bugpoint-minimized.sv:49:8: Parameter type pin value isn't a type: Param 'data_t' of 'rand_id_queue'
                                     : ... note: In instance 'tb_axi_xbar'
   49 |       .data_t   (axi_driver_t::ax_beat_t),
      |        ^~~~~~
        ... See the manual at https://verilator.org/verilator_doc.html?v=5.037 for more assistance.
%Error: Internal Error: sv-bugpoint-minimized.sv:49:32: ../V3Ast.h:2656: AstNode is not of expected type, but instead has type 'DOT'
                                                      : ... note: In instance 'tb_axi_xbar'
   49 |       .data_t   (axi_driver_t::ax_beat_t),
      |                                ^~~~~~~~~
                        ... This fatal error may be caused by the earlier error(s); resolve those first.
```

Can you attach an example that shows the issue?  (Must be openly licensed, completely self-contained and self-checking so can directly run what you provide. Ideally use test_regress format, see https://veripool.org/guide/latest/contributing.html?highlight=test_regress#reporting-bugs

```
package rand_id_queue_pkg;
class rand_id_queue #(
  type          data_t = logic,
  int unsigned  ID_WIDTH = 0
);
endclass
endpackage
package axi_test;
  class axi_ax_beat #(
    parameter int unsigned AW = 32,
    parameter int unsigned IW = 8 ,
    parameter int unsigned UW = 1
  );
    rand logic [IW-1:0] ax_id     = '0;
    rand logic [AW-1:0] ax_addr   = '0;
    logic [7:0]         ax_len    = '0;
    logic [2:0]         ax_size   = '0;
    logic [1:0]         ax_burst  = '0;
    logic               ax_lock   = '0;
    logic [3:0]         ax_cache  = '0;
    logic [2:0]         ax_prot   = '0;
    rand logic [3:0]    ax_qos    = '0;
    logic [3:0]         ax_region = '0;
    logic [5:0]         ax_atop   = '0;  
    rand logic [UW-1:0] ax_user   = '0;
  endclass
  class axi_driver #(
    parameter int unsigned AW = 32  ,
    parameter int unsigned DW = 32  ,
    parameter int unsigned IW = 8   ,
    parameter int unsigned UW = 1   ,
    parameter time         TA = 0ns ,  
    parameter time         TT = 0ns    
  );
    typedef axi_ax_beat #(.AW(AW), .IW(IW), .UW(UW)) ax_beat_t;
  endclass
  class axi_rand_slave #(
    parameter int   AW = 32,
    parameter int   DW = 32,
    parameter int   IW = 8,
    parameter int   UW = 1,
    parameter time  TA = 0ps,
    parameter time  TT = 0ps
  );
    typedef axi_test::axi_driver #(
      .AW(AW), .DW(DW), .IW(IW), .UW(UW), .TA(TA), .TT(TT)
    ) axi_driver_t;
    typedef rand_id_queue_pkg::rand_id_queue #(
      .data_t   (axi_driver_t::ax_beat_t),
      .ID_WIDTH (IW)
    ) rand_ax_beat_queue_t;
  endclass
endpackage
module tb_axi_xbar #(
  parameter int unsigned TbNumMasters        = 32'd6,
  parameter int unsigned TbAxiIdWidthMasters = 32'd5,
  parameter int unsigned TbAxiDataWidth      = 32'd64       
);
  localparam time ApplTime =  2ns;
  localparam time TestTime =  8ns;
  localparam int unsigned TbAxiIdWidthSlaves =  TbAxiIdWidthMasters + $clog2(TbNumMasters);
  localparam int unsigned TbAxiAddrWidth     =  32'd32;
  localparam int unsigned TbAxiUserWidth     =  5;
  typedef axi_test::axi_rand_slave #(
    .AW ( TbAxiAddrWidth     ),
    .DW ( TbAxiDataWidth     ),
    .IW ( TbAxiIdWidthSlaves ),
    .UW ( TbAxiUserWidth     ),
    .TA ( ApplTime           ),
    .TT ( TestTime           )
  ) axi_rand_slave_t;
endmodule
```

What 'verilator' command line do we use to run your example?

```
verilator --top-module tb_axi_xbar --binary --timing sv-bugpoint-minimized.sv -Wno-fatal
```

What 'verilator --version' are you using?  Did you try it with the git master version?

`v5.037-g49fe12963`

What OS and distribution are you using?

`Linux 5.14.0-570.19.1.el9_6.x86_64`

May we assist you in trying to fix this in Verilator yourself?

(Please avoid attaching screenshots that show text - you can convert images to text using e.g. https://ocr.space)

solomatnikov, 2025-06-21:
Full trace:

```
verilator --debug --gdbbt --top-module tb_axi_xbar --binary --timing sv-bugpoint-minimized.sv -Wno-fatal
export VERILATOR_ROOT='/wss/tools/verilator/v5.037-g49fe12963/share/verilator'
No stack.
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
Starting Verilator 5.037 devel rev v5.036-152-g49fe12963
- Verilator.cpp:651:  Option --verilate: Start Verilation
- V3File.cpp:234:        --check-times failed: different command line
- V3ParseImp.cpp:301: parseFile: verilated_std_waiver
- V3PreShell.cpp:90:  Preprocessing /wss/tools/verilator/v5.037-g49fe12963/share/verilator/include/verilated_std_waiver.vlt
- V3PreShell.cpp:143:     Reading /wss/tools/verilator/v5.037-g49fe12963/share/verilator/include/verilated_std_waiver.vlt
- V3ParseImp.cpp:383: Lexing /wss/tools/verilator/v5.037-g49fe12963/share/verilator/include/verilated_std_waiver.vlt
- V3ParseImp.cpp:301: parseFile: verilated_std
- V3PreShell.cpp:90:  Preprocessing /wss/tools/verilator/v5.037-g49fe12963/share/verilator/include/verilated_std.sv
- V3PreShell.cpp:143:     Reading /wss/tools/verilator/v5.037-g49fe12963/share/verilator/include/verilated_std.sv
- V3ParseImp.cpp:383: Lexing /wss/tools/verilator/v5.037-g49fe12963/share/verilator/include/verilated_std.sv
- V3ParseImp.cpp:301: parseFile: sv-bugpoint-minimized
- V3PreShell.cpp:90:  Preprocessing sv-bugpoint-minimized.sv
- V3PreShell.cpp:143:     Reading sv-bugpoint-minimized.sv
- V3ParseImp.cpp:369: Writing all preprocessed output to obj_dir/Vtb_axi_xbar__inputs.vpp
- V3ParseImp.cpp:383: Lexing sv-bugpoint-minimized.sv
- V3LinkCells.cpp:210:Link --top-module: MODULE 0x12aef60 <e1583#> {e54ai} u1=0x12b4c70  tb_axi_xbar  L0 [NONE]
- V3LinkLevel.cpp:42: modSortByLevel()
- V3Ast.cpp:1347:     Dumping obj_dir/Vtb_axi_xbar_002_cellsort.tree
- V3LinkDot.cpp:4647: linkDotPrimary:
- V3LinkJump.cpp:384: linkJump:
- V3Ast.cpp:1347:     Dumping obj_dir/Vtb_axi_xbar_007_linkjump.tree
- V3LinkInc.cpp:370:  linkIncrements:
- V3Ast.cpp:1347:     Dumping obj_dir/Vtb_axi_xbar_008_linkinc.tree
- V3Param.cpp:1559:   param:
%Error: sv-bugpoint-minimized.sv:49:8: Parameter type pin value isn't a type: Param 'data_t' of 'rand_id_queue'
                                     : ... note: In instance 'tb_axi_xbar'
-node: PIN 0x12bd670 <e2249> {e49ai}  data_t ->UNLINKED [.n]
   49 |       .data_t   (axi_driver_t::ax_beat_t),
      |        ^~~~~~
        ... See the manual at https://verilator.org/verilator_doc.html?v=5.037 for more assistance.
%Error: Internal Error: sv-bugpoint-minimized.sv:49:32: ../V3Ast.h:2656: AstNode is not of expected type, but instead has type 'DOT'
                                                      : ... note: In instance 'tb_axi_xbar'
-node: DOT 0x12b6770 <e1570> {e49bg} @dt=0@ [::]
   49 |       .data_t   (axi_driver_t::ax_beat_t),
      |                                ^~~~~~~~~
                        ... This fatal error may be caused by the earlier error(s); resolve those first.
- V3Ast.cpp:1347:     Dumping obj_dir/Vtb_axi_xbar_990_final.tree
- V3StatsReport.cpp:231:statsReport:
%Error: Internal Error: Aborting since under --debug

Program received signal SIGABRT, Aborted.
0x0000155554e8bedc in __pthread_kill_implementation () from /lib64/libc.so.6
#0  0x0000155554e8bedc in __pthread_kill_implementation () from /lib64/libc.so.6
#1  0x0000155554e3eb46 in raise () from /lib64/libc.so.6
#2  0x0000155554e28833 in abort () from /lib64/libc.so.6
#3  0x000000000048e393 in V3Error::vlAbort () at ../V3Error.cpp:369
#4  0x000000000048fa3b in V3ErrorGuarded::vlAbortOrExit (this=this@entry=0x1229aa0 <V3Error::s()::s_s>) at ../V3Error.cpp:98
#5  0x0000000000492100 in V3ErrorGuarded::v3errorEndGuts (this=this@entry=0x1229aa0 <V3Error::s()::s_s>, sstr=..., extra="__WARNBEGIN__", ' ' <repeats 30 times>, ": __WARNEND__... note: In instance 'tb_axi_xbar'", fileline=fileline@entry=0x12ade30) at ../V3Error.cpp:278
#6  0x00000000004924c6 in V3ErrorGuarded::v3errorEnd (this=0x1229aa0 <V3Error::s()::s_s>, sstr=..., extra="__WARNBEGIN__", ' ' <repeats 30 times>, ": __WARNEND__... note: In instance 'tb_axi_xbar'", fileline=fileline@entry=0x12ade30) at ../V3Error.cpp:138
#7  0x000000000049267d in V3Error::v3errorEnd (sstr=..., extra="__WARNBEGIN__", ' ' <repeats 30 times>, ": __WARNEND__... note: In instance 'tb_axi_xbar'", fileline=fileline@entry=0x12ade30) at ../V3Error.cpp:383
#8  0x000000000049ba5d in FileLine::v3errorEnd (this=0x12ade30, sstr=..., extra="... note: In instance 'tb_axi_xbar'") at ../V3FileLine.cpp:451
#9  0x00000000005b56de in AstNode::v3errorEnd (this=this@entry=0x12b6770, str=...) at ../V3Ast.cpp:1466
#10 0x00000000005b5b7d in AstNode::v3errorEndFatal (this=this@entry=0x12b6770, str=...) at ../V3Ast.cpp:1470
#11 0x000000000043223b in AstNode::privateAs<AstNodeDType, AstNode*> (nodep=nodep@entry=0x12b6770) at ../V3Ast.h:2656
#12 0x0000000000b4700c in ParamProcessor::deepCloneModule (this=this@entry=0x7fffffffcc50, srcModp=srcModp@entry=0x12982f0, ifErrorp=ifErrorp@entry=0x12bd5a0, paramsp=paramsp@entry=0x12bd670, newname="rand_id_queue__I8", ifaceRefRefs=std::__debug::deque with 0 elements) at ../V3Param.cpp:684
#13 0x0000000000b47646 in ParamProcessor::moduleFindOrClone (this=this@entry=0x7fffffffcc50, srcModp=0x12982f0, ifErrorp=ifErrorp@entry=0x12bd5a0, paramsp=paramsp@entry=0x12bd670, newname="rand_id_queue__I8", ifaceRefRefs=std::__debug::deque with 0 elements) at ../V3Param.cpp:702
[... cut ...]
#29 0x0000000000b208ab in V3Param::param (rootp=0x1240be0) at ../V3Param.cpp:1560
#30 0x00000000005aa4e4 in process () at ../V3Global.h:151
#31 0x00000000005abe7f in verilate (argString="--debug --gdbbt --top-module tb_axi_xbar --binary --timing sv-bugpoint-minimized.sv -Wno-fatal") at ../Verilator.cpp:708
#32 0x00000000005acf83 in main (argc=<optimized out>, argv=0x7fffffffd748) at ../Verilator.cpp:854
A debugging session is active.

	Inferior 1 [process 188572] will be killed.

Quit anyway? (y or n) [answered Y; input not from terminal]
```

wsnyder, 2025-06-22:
Thanks, the example is ok as is.  The issue is V3Param wants the `::` should be resolved in the first V3LinkDot, but it is not and cannot be completely resolved (until later) because of the parameter.  Probably it will need to be half-converted to a class reference in the first V3LinkDot. Needs more thought later.

wsnyder, 2025-11-03:
As to comment in #6627:

The first LinkDot will set the :: references to modules that are not parameterized, and provide other hints for V3Param to work.
Then V3Param runs and duplicates the modules.  It attempts to resolve types as needed for parameterization.
The second LinkDot Param will then resolve the parameterized :: references.

It's likely that improvements to the first LinkDot + V3Param can resolve this. Hopefully it won't suggest needing a fundamental rewrite to remove the multiple LinkDots (which are unfortunate result of how Verilator evolved as the language standard became more and more complicated).

nohous, 2025-11-03:
It's basically about interleaved substitution, right? Since the #6627 's TEST3 passes, it suggests that maybe introducing some temporary placeholder AstNode in the first LinkDot might do the trick? It feels hacky though. Should I try to get down the rabbit hole?

wsnyder, 2025-11-03:
The first LinkDot in other cases sets the appropriate classOrPackageNodep to pass to later stages, so presumably similar would work here.  (Or, it was intended to do so already, and doesn't for some reason, I'm not sure.)