verilator/verilator #6106: co-routine issue building sequencer

joaocarvalho00, 2025-06-21:
I'm having issues running verilator with a basic UVM environment due to co-routine issues on the sequencer.
I'm on ubuntu 22.04, verilator v5.036, gcc 13.3

I've created a "hello world" example RTL + UVM TB at "https://github.com/joaocarvalho00/tiny_risc" that works by calling "make build"

I've followed the examples at https://github.com/MikeCovrado/GettingVerilatorStartedWithUVM and downloaded the UVM library from antmicro. I can run this example successfully but when I try to run my own example with the following flags I get this error:

I've tried removing --trace, adding std=g++20. removing UVM_NO_DPI, adding --top and nothing worked.

```
export UVM_HOME="/home/joaocarvalho00/Projects/verilator-verification-features-tests/uvm/uvm-2017/src"
UVM_PKG="${UVM_HOME}/uvm_pkg.sv"

verilator \
	--binary \
        --build \
        --cc \
        --timing \
        --exe \
	    --trace \
        --Mdir verilator_obj_dir \
        -j 1 \
	    +incdir+${UVM_HOME} \
        +incdir+src \
        +incdir+tb \
	    ${UVM_PKG} \
        +define+UVM_REPORT_DISABLE_FILE_LINE \
        +define+SVA_ON \
        +define+UVM_NO_DPI \
        tb/tb_cpu_pkg.sv \
        tb/tb_cpu_if.sv \
        src/cpu.sv \
        tb/tb_cpu.sv \
        -Wno-DECLFILENAME    \
        -Wno-CONSTRAINTIGN   \
        -Wno-MISINDENT       \
        -Wno-VARHIDDEN       \
        -Wno-WIDTHTRUNC      \
        -Wno-CASTCONST       \
        -Wno-WIDTHEXPAND     \
        -Wno-UNDRIVEN        \
        -Wno-UNUSEDSIGNAL    \
        -Wno-UNUSEDPARAM     \
        -Wno-ZERODLY         \
        -Wno-SYMRSVDWORD     \
        -Wno-CASEINCOMPLETE  \
        -Wno-SIDEEFFECT      \
        -Wno-REALCVT
```

Vuvm_pkg_uvm_pkg__03a__03auvm_sequencer__Tz117_TBz117__Vclpkg__DepSet_h94236e61__0.cpp:29:17: error: ‘co_await’ cannot be used in a function with a deduced return type
   29 |                 co_await VL_NULL_CHECK(Vuvm_pkg_uvm_pkg__03a__03auvm_sequencer_param_base__pi73::__PVT__m_req_fifo, "/home/joaocarvalho00/Projects/verilator-verification-features-tests/uvm/uvm-2017/src/seq/uvm_sequencer.svh", 286)
      |                 ^~~~~~~~
Vuvm_pkg_uvm_pkg__03a__03auvm_sequencer__Tz117_TBz117__Vclpkg__DepSet_h94236e61__0.cpp: In member function ‘virtual VlCoroutine Vuvm_pkg_uvm_pkg__03a__03auvm_sequencer__Tz117_TBz117::__VnoInFunc_item_done(VlProcessRef, Vuvm_pkg__Syms*, VlClassRef<Vuvm_pkg_tb_cpu_pkg__03a__03atb_cpu_seq_item>)’:
Vuvm_pkg_uvm_pkg__03a__03auvm_sequencer__Tz117_TBz117__Vclpkg__DepSet_h94236e61__0.cpp:50:1: warning: no return statement in function returning non-void [-Wreturn-type]
   50 | }

wsnyder, 2025-06-21:
We are not supporting UVM yet (#1538).

If you want to reduce this case to one showing the issue with e.g. sv-bugpoint (see [Reporting Verilator Bugs](https://verilator.org/guide/latest/contributing.html?highlight=test_regress#reporting-bugs) and section on use of sv-bugpoint), that is completely self contained and does not use any UVM beyond defining a few uvm_ classes, and is reasonably small (100 lines at most), we can look into it.

Alternatively it would be very helpful if you debug Verilator yourself, e.g. look at what causes the C++ to be wrong, then look at the --debug .tree files to see where it goes wrong (likely V3Fork).  We'll still need a small test case to prove the fix though.

solomatnikov, 2025-06-21:
Using `sv-bugpoint` for such cases is painful b/c it immediately hits unrelated verilator bugs:

```
%Error: Internal Error: attempt1.sv:7:33: ../V3LinkDot.cpp:2756: Pin not under instance?
```

wsnyder, 2025-06-23:
Thanks for reducing.

wsnyder, 2025-06-26:
The problem is the coroutine logic doesn't like the lambda made by below.  A workaround is to change the code as shown.

```
    function void uvm_sequencer::item_done( int item = null);
       REQ t;
-      if (m_req_fifo.try_get(t) == 0) begin
+      bit res;
+      res = m_req_fifo.try_get(t);
+      if (res == 0) begin
       end
```

It's trivial to change the C++ lambda to have return types, but hand-trying that it just results in a different coroutine-related error.

@kboronski-ant if you get a chance, thanks.

joaocarvalho00, 2025-08-10:
Hi,

I changed the uvm_sequencer.svh to have the code you mention there but I still get the same error: 

```
function void uvm_sequencer::item_done(RSP item = null);
  REQ t;
  bit res;

  // Set flag to allow next get_next_item or peek to get a new sequence_item
  sequence_item_requested = 0;
  get_next_item_called = 0;
  
  
  res = m_req_fifo.try_get(t);
  if (res == 0) begin
    uvm_report_fatal("SQRBADITMDN", {"Item_done() called with no outstanding requests.",
      " Each call to item_done() must be paired with a previous call to get_next_item()."});
  end else begin
    m_wait_for_item_sequence_id = t.get_sequence_id();
    m_wait_for_item_transaction_id = t.get_transaction_id();
  end
  
  if (item != null) begin
    seq_item_export.put_response(item);
  end

  // Grant any locks as soon as possible
  grant_queued_locks();
endfunction
```

solomatnikov, 2025-08-30:
Now I get:

```
%Error: t/t_issue6106.v:55:12: Function Argument 'class{}tb_cpu_seq_item' cannot be assigned to non-class 'int'
                             : ... note: In instance 'tb_cpu'
   55 |     get_ap.write( t );
      |            ^~~~~
```

Not clear why.

wsnyder, 2025-08-30:
Means that you probably worked around this and now are hitting #4920 or #4497 .  Remove all parameters to be certain.

solomatnikov, 2025-08-30:
> Means that you probably worked around this and now are hitting [#4920](https://github.com/verilator/verilator/issues/4920) or [#4497](https://github.com/verilator/verilator/issues/4497) . Remove all parameters to be certain.

Idk what you mean but when I try to get around this by:

```
package uvm_pkg;
typedef enum
{
  UVM_PORT ,
  UVM_IMPLEMENTATION
} uvm_port_type_e;
virtual class uvm_void;
virtual class uvm_transaction;
  extern function int get_transaction_id();
endclass
function int uvm_transaction::get_transaction_id();
endfunction
virtual class uvm_component;
  extern function new (string name, int parent);
endclass : uvm_component
function uvm_component::new (string name, int parent);
endfunction  
virtual class uvm_tlm_if_base #(type T1=int, type T2=int);
  virtual function void write( input int t );
  endfunction
endclass
const int UVM_UNBOUNDED_CONNECTIONS = -1;
virtual class uvm_port_base #(type IF=uvm_void) extends IF;
  function new (string name, int parent, int port_type,
                int min_size=0,
                int max_size=1);
  endfunction
endclass
class uvm_analysis_port # (type T = int)
  extends uvm_port_base # (uvm_tlm_if_base #(T,T));
  function new (string name, int parent);
    super.new (name, parent, UVM_PORT, 0, UVM_UNBOUNDED_CONNECTIONS);  
  endfunction
endclass
class uvm_analysis_imp #(type T=int, type IMP=int)
  extends uvm_port_base #(uvm_tlm_if_base #(T,T));
  local IMP m_imp; 
  function new (string name, int imp); 
    super.new (name, imp, UVM_IMPLEMENTATION, 1, 1); 
  endfunction
  function void write (input int t);
    m_imp.write (t);
  endfunction
endclass
virtual class uvm_tlm_fifo_base #(type T=int) extends uvm_component;
  uvm_analysis_port #(T) get_ap;
  function new(string name, int parent = null);
    super.new(name, parent);
  endfunction
endclass
class uvm_tlm_fifo #(type T=int) extends uvm_tlm_fifo_base #(T);
  function new(string name, int parent = null, int size = 1);
    super.new(name, parent);
  endfunction
  virtual function bit try_get( output T t );
    get_ap.write( t );
  endfunction
endclass
class uvm_sequence_item extends uvm_transaction;
endclass
virtual
class uvm_sequencer_base extends uvm_component;        
  protected int                 m_wait_for_item_sequence_id,
                                m_wait_for_item_transaction_id;
  extern function new (string name, int parent);
endclass
function uvm_sequencer_base::new (string name, int parent);
  super.new(name, parent);
endfunction
virtual
 class uvm_sequencer_param_base #(type REQ = int,
                                 type RSP = int) extends uvm_sequencer_base;
  extern function new (string name, int parent);
    uvm_tlm_fifo #(REQ) m_req_fifo;
endclass
function uvm_sequencer_param_base::new (string name, int parent);
  super.new(name, parent);
endfunction
class uvm_sequencer #(type REQ= int, RSP= int)
                                   extends uvm_sequencer_param_base #(REQ, RSP); 
  extern function new (string name, int parent=null);
  extern virtual function void item_done     ( int item = null);
endclass
function uvm_sequencer::new (string name, int parent=null);
  super.new(name, parent);
endfunction
function void uvm_sequencer::item_done( int item = null);
  REQ t;
  if (m_req_fifo.try_get(t) == 0) begin
  end else begin
    m_wait_for_item_transaction_id = t.get_transaction_id();
  end
endfunction
endpackage
import uvm_pkg::*;
interface cpu_if(input logic clk);       
endinterface : cpu_if;
package tb_cpu_pkg;
     class tb_cpu_seq_item extends uvm_sequence_item;
endclass : tb_cpu_seq_item
class tb_cpu_sequencer extends uvm_sequencer #(tb_cpu_seq_item, tb_cpu_seq_item); 
   function new(string name, int parent);
      super.new(name, parent);
   endfunction: new
endclass : tb_cpu_sequencer
     class tb_cpu_scoreboard extends uvm_component; 
    uvm_analysis_imp #(tb_cpu_seq_item, tb_cpu_scoreboard) item_collected_export;
    virtual cpu_if vif;
    function new(string name, int parent);
        super.new(name, parent);
    endfunction
    virtual task write( int t);
        @(posedge vif.clk);
    endtask
endclass : tb_cpu_scoreboard
endpackage : tb_cpu_pkg
module tb_cpu;
    cpu_if vif(clk);
endmodule // tb_cpu
```

`verilator` is confused:

```
%Error: t/t_issue6106_2.v:94:1: syntax error, unexpected endpackage
   94 | endpackage
      | ^~~~~~~~~~
        ... See the manual at https://verilator.org/verilator_doc.html?v=5.039 for more assistance.
%Error: t/t_issue6106_2.v:116:1: syntax error, unexpected endpackage
  116 | endpackage : tb_cpu_pkg
      | ^~~~~~~~~~
%Error: Cannot continue
```

solomatnikov, 2025-09-02:
New test case:

```
package uvm_pkg;
typedef enum
{
  UVM_PORT ,
  UVM_IMPLEMENTATION
} uvm_port_type_e;
virtual class uvm_component;
  extern function new (string name, uvm_component parent);
endclass : uvm_component
class uvm_root extends uvm_component;
	extern protected function new ();
endclass
function uvm_root::new();
  super.new("__top__", null);
endfunction
function uvm_component::new (string name, uvm_component parent);
endfunction  
virtual class uvm_tlm_if_base #(type T1=int, type T2=int);
  virtual function void write( input T1 t );
  endfunction
endclass
const int UVM_UNBOUNDED_CONNECTIONS = -1;
virtual class uvm_port_base #(type IF= int) extends IF;
  function new (string name,
                uvm_component parent, int port_type,
                int min_size,
                int max_size);
  endfunction
endclass
class uvm_analysis_port # (type T = int)
  extends uvm_port_base # (uvm_tlm_if_base #(T,T));
  function new (string name, uvm_component parent);
    super.new (name, parent, UVM_PORT, 0, UVM_UNBOUNDED_CONNECTIONS);  
  endfunction
endclass
class uvm_analysis_imp #(type T=int, type IMP=int)
  extends uvm_port_base #(uvm_tlm_if_base #(T,T));
  local IMP m_imp; 
  function new (string name, IMP imp); 
    super.new (name, imp, UVM_IMPLEMENTATION, 1, 1); 
  endfunction
  function void write (input T t);
    m_imp.write (t);
  endfunction
endclass
virtual class uvm_tlm_fifo_base #(type T=int) extends uvm_component;
  uvm_analysis_port #(T) get_ap;
  function new(string name, uvm_component parent);
    super.new(name, parent);
  endfunction
endclass
class uvm_tlm_fifo #(type T=int) extends uvm_tlm_fifo_base #(T);
  local mailbox #( T ) m;
  function new(string name, uvm_component parent, int size);
    super.new(name, parent);
  endfunction
  virtual function bit try_get( output T t );
    if( !m.try_get( t ) ) begin
    end
    get_ap.write( t );
  endfunction
endclass
typedef int uvm_config_seq;
virtual
class uvm_sequencer_base extends uvm_component;
  extern function new (string name, uvm_component parent);
endclass
[... 72 lines cut ...]
Vtb_cpu_uvm_pkg__03a__03auvm_sequencer__Tz1_TBz1__Vclpkg__DepSet_he0083cb8__0.cpp:24:1: warning: no return statement in function returning non-void [-Wreturn-type]
   24 | }
      | ^
```

VCS:
```
Warning-[TEIF] Task enabled inside a function
testbench.sv, 43
uvm_pkg, "tb_cpu_pkg::\tb_cpu_scoreboard::write (this.m_imp, t);"
  Task 'tb_cpu_scoreboard::write' is enabled inside function 
  'uvm_analysis_imp#(tb_cpu_pkg::tb_cpu_seq_item,tb_cpu_pkg::tb_cpu_scoreboard)::write',
  it may bring delays into the function.

Error-[SV-DOSIF] Delay or synchronization in function
testbench.sv, 42
uvm_pkg, "tb_cpu_pkg::\tb_cpu_scoreboard::write (this.m_imp, t);"
  Function 'uvm_pkg_uvm_analysis_imp_549398629462903228_78::write' has illegal
  use of delay or synchronization
  The uses can be wait, delay, clocking block assign, fork-join and other task
  calls with delays

1 warning
1 error
```

solomatnikov, 2025-09-24:
After #6385:

```
%Error: uvm-verilator/src/tlm1/uvm_analysis_port.svh:117:11: Functions cannot invoke tasks (IEEE 1800-2023 13.4)
                                                                                               : ... note: In instance 'tb_cpu.dut'
  117 |     m_imp.write (t);
      |           ^~~~~
        uvm-verilator/src/tlm1/uvm_tlm.svh:34:1: ... note: In file included from 'uvm_tlm.svh'
        uvm-verilator/src/uvm_pkg.sv:34:1: ... note: In file included from 'uvm_pkg.sv'
                                                                                               : ... Suggest make caller 'function write' a task
  116 |   function void write (input T t);
      |                 ^~~~~
                                                                                               : ... Or, suggest make called 'task write' a function void
   26 |     virtual task write(tb_cpu_seq_item t);
      |                  ^~~~~
```

wsnyder, 2026-02-20:
@ArturBieniek4 might you or one of your colleagues get a chance to look at this?, thanks.