verilator/verilator #6109: Why does --hierarchical regenerate all blocks on any source change?

Nofal475, 2025-06-23:
I have noticed that when I build with --hierarchical, Verilator produces a Vprefix_top.hier file that lists every RTL file as a dependency for each generated hierarchical block.
As a result changing any single source file triggers a complete re-verilation of all hier blocks, eliminating the incremental build benefit I was hoping for.

My expectation was that if a leaf block ports and parameters are unchanged, verilator to only list the files that directly define the block.

Is this intentional? Required for correctness ?

Thanks for any guidance.

Nofal475, 2025-06-23:
`
VM_HIER_INPUT_FILES := \
	/test_project/counter_16bit_A.sv \
	/test_project/counter_16bit_B.sv \
	/test_project/counter_top.sv \

Vcounter_16bit_B/counter_16bit_B.sv Vcounter_16bit_B/Vcounter_16bit_B.mk: $(VM_HIER_INPUT_FILES) ... 

Vcounter_16bit_A/counter_16bit_A.sv Vcounter_16bit_A/Vcounter_16bit_A.mk: $(VM_HIER_INPUT_FILES) ...
`

where as I expected it to be 

`
Vcounter_16bit_B/counter_16bit_B.sv Vcounter_16bit_B/Vcounter_16bit_B.mk: /test_project/counter_16bit_B.sv ... 

Vcounter_16bit_A/counter_16bit_A.sv Vcounter_16bit_A/Vcounter_16bit_A.mk: /test_project/counter_16bit_A.sv ...

`
BTW using verilator 5.032

Nofal475, 2025-06-23:
Thank you @wsnyder ! 

Just for clarification, if we manually manage dependencies, for example, by ensuring that each hierarchical block .mk file only depends on its own RTL source and a shared file like common_defines.svh that contains all relevant macros — would that allow us to safely get incremental rebuilds?

wsnyder, 2025-06-23:
No, let make invoke verilator but V3File::checkTimes, when timestamps differ, will then see if the hashes of the compared files are the same and if so consider them identical.  (Good to still have a mtime check as skips needing the checksum.)

wsnyder, 2025-07-17:
@Nofal475 wasn't sure from this thread if you are expecting to work on this or not.  Not trying to rush, just curious.