verilator/verilator #6120: Support dotted references in bind

solomatnikov, 2025-06-25:
Thanks for taking the time to report this.

What would you like added/supported?

dotted references in bind

What 'verilator --version' are you using?  Did you try it with the git master version?

`Verilator 5.037 devel rev v5.036-162-g51545b36c`

Can you attach an example that runs on other simulators?  (Must be openly licensed, completely self-contained and self-checking so can directly run what you provide. Ideally use test_regress format, see https://veripool.org/guide/latest/contributing.html?highlight=test_regress#reporting-bugs

```
git diff
diff --git a/test_regress/t/t_bind2.v b/test_regress/t/t_bind2.v
index 273e21b31..308da8537 100644
--- a/test_regress/t/t_bind2.v
+++ b/test_regress/t/t_bind2.v
@@ -32,22 +32,6 @@ module t (/*AUTOARG*/
                           // Inputs
                           .clk                  (clk));
 
-   //Binding i_targetmod to mycheck --instantiates i_mycheck inside i_targetmod
-   //param1 not over-riden (as mycheck)                 (=> 0x31)
-   //param2 explicitly bound to targetmod value (=> 0x22)
-   //param3 explicitly bound to top value                       (=> 0x13)
-   //p1 implictly bound (.*), takes value from targetmod        (=> 0x04)
-   //p2 explictly bound to targetmod                                            (=> 0x05)
-   //p3 explictly bound to top                                                          (=> 0x03)
-
-   // Alternative unsupported form is i_targetmod
-   bind targetmod mycheck
-     #(
-       .param2(param2),
-       .param3(param3)
-       )
-   i_mycheck (.p2(p2), .p3(p3), .*);
-
 endmodule
 
 module targetmod (input clk);
@@ -91,3 +75,14 @@ module mycheck (/*AUTOARG*/
       $finish;
    end
 endmodule
+
+   //Binding i_targetmod to mycheck --instantiates i_mycheck inside i_targetmod
+   //param1 not over-riden (as mycheck)                 (=> 0x31)
+   //param2 explicitly bound to targetmod value (=> 0x22)
+   //param3 explicitly bound to top value                       (=> 0x13)
+   //p1 implictly bound (.*), takes value from targetmod        (=> 0x04)
+   //p2 explictly bound to targetmod                                            (=> 0x05)
+   //p3 explictly bound to top                                                          (=> 0x03)
+
+   // Alternative unsupported form is i_targetmod
+   bind t.i_targetmod mycheck i_mycheck (.p2(p2), .p3(p3), .*);
```

```
%Error: t/t_bind2.v:88:10: syntax error, unexpected '.', expecting IDENTIFIER or IDENTIFIER-for-instance or ':' or randomize
   88 |    bind t.i_targetmod mycheck i_mycheck (.p2(p2), .p3(p3), .*);
      |          ^
```

**Compiles with VCS**

May we assist you in trying to fix this in Verilator yourself?

wsnyder, 2025-06-25:
Made another test, shown below.  Not sure how soon can get to implementing this, as always pull welcome.

```
`define stop $stop
`define checkh(gotv,expv) do if ((gotv) !== (expv)) begin $write("%%Error: %s:%0d:  got='h%x exp='h%x\n", `__FILE__,`__LINE__, (gotv), (expv)); `stop; end while(0);

module t (/*AUTOARG*/);

   // Identifiers are relative to bind target
   bind moda: t.u_a1, t.u_a2 mycheck #(.CHECK('h90)) u_check();

   bind u_a1.u_b1 mycheck #(.CHECK('h11)) u_check();
   bind u_a1.u_b2 mycheck #(.CHECK('h12)) u_check();
   bind u_a2.u_b1 mycheck #(.CHECK('h21)) u_check();
   bind u_a2.u_b2 mycheck #(.CHECK('h22)) u_check();

   moda #(.VAL('h10)) u_a1();
   moda #(.VAL('h20)) u_a2();

   initial begin
     `checkh(u_a1.u_check.CHECK, 'h90);
     `checkh(u_a2.u_check.CHECK, 'h90);
     `checkh(u_a1.u_b1.u_check.CHECK, 'h11);
     `checkh(u_a1.u_b2.u_check.CHECK, 'h12);
     `checkh(u_a2.u_b1.u_check.CHECK, 'h21);
     `checkh(u_a2.u_b2.u_check.CHECK, 'h22);
      $write("*-* All Finished *-*\n");
      $finish;
   end

endmodule

module moda;
   parameter VAL = 0;
   modb #(.VAL(VAL+1)) u_b1();
   modb #(.VAL(VAL+2)) u_b2();
endmodule

module modb;
   parameter VAL = 0;
endmodule

module mycheck;
   parameter CHECK = 0;
endmodule
```

wsnyder, 2026-11-09:
The dotted references now are parsed, but still otherwise unsupported.