verilator/verilator #6134: Support for array.sum() with (...) Inside Constraint Blocks?

raulbehl, 2025-06-27:
I wanted to check if Verilator currently supports the use of array.sum() with (...) inside a constraint block.

From a quick look at the `verilator_types.h` file, I can see that the `r_sum` method appears to support a with function:
```c
template <typename T_Func>
WithFuncReturnType<T_Func> r_sum(T_Func with_func) const {
    WithFuncReturnType<T_Func> out = WithFuncReturnType<T_Func>(0);
    IData index = 0;
    for (const auto& i : m_deque) out += with_func(index++, i);
    return out;
}
```

However, when I try to use it in the following SystemVerilog constraint:
```verilog
constraint duplicates_only_c {
  foreach(array[i]) {
    array.sum() with (int'(item == array[i])) == 1;
  }
}
```

I receive the following error:
```text
%Error: unique_only.sv:25:19: 'with' not legal on this method
                                                                                                                                                         : ... note: In instance 'unique_only_top'
   25 |       array.sum() with (int'(item==array[i]))== 1;
      |                   ^~~~
```

Is the functionality supported?

raulbehl, 2025-06-27:
Thank you for the clarification @wsnyder. Unfortunately, I still can't find my way around verilator code base but can try using AI companion to help support this. Could you guide on where should I start looking into?

raulbehl, 2025-09-15:
@wsnyder Is this suppose to be supported? I am on version `Verilator 5.041 devel rev v5.040-121-gdca555b6d` but I still get the following: 

```
%Warning-CONSTRAINTIGN: top.sv:33:11: Unsupported: randomizing this expression, treating as state
   33 |     array.sum() with (int'(item == array[i])) == 1;
      |           ^~~
                        top.sv:13:1:
                        ... For warning description see https://verilator.org/warn/CONSTRAINTIGN?v=5.041
                        ... Use "/* verilator lint_off CONSTRAINTIGN */" and lint_on around source to disable this message.
%Error: Exiting due to 1 warning(s)
```

wsnyder, 2025-09-16:
Sorry, the problem was misinterpreted as `.sum() with` outside randomize, instead of for randomize constraints.  Can you please open a new issue?  Also would appreciate if the code example is compliable as then likely to check it better ;), thanks.