verilator/verilator #6146: Unsupported: Assert not allowed under another assert for assert(std::randomize() with

solomatnikov, 2025-07-01:
Thanks for taking the time to report this.

Can you attach an example that shows the issue?  (Must be openly licensed, completely self-contained and self-checking so can directly run what you provide. Ideally use test_regress format, see https://veripool.org/guide/latest/contributing.html?highlight=test_regress#reporting-bugs

```
%Error-UNSUPPORTED: base_test_wrapper.sv:1380:2: Unsupported: Assert not allowed under another assert
 1380 |  assert(std::randomize(dis) with
      |  ^~~~~~
```

```
 assert(std::randomize(dis) with
 {
  dis[5] == 0; 
  dis[19] == 0; 
  $countones({~dis[32-1:16] & bif,~dis[16-1:0]}) > 1;
 });
```

What 'verilator' command line do we use to run your example?

`verilator --binary --sv --assert -Wno-fatal`

What 'verilator --version' are you using?  Did you try it with the git master version?

`Verilator 5.037 devel rev v5.036-189-gcd0f35fe6`

What OS and distribution are you using?

`Linux 4.18.0-553.22.1.el8_10.x86_64`

May we assist you in trying to fix this in Verilator yourself?

(Please avoid attaching screenshots that show text - you can convert images to text using e.g. https://ocr.space)

wsnyder, 2025-07-10:
As you noted that example doesn't fail, and even if countones gets fixed the first example isn't complete, so we'll need something that self-contain-fails to debug, thanks.

wsnyder, 2025-07-21:
It is an assert under an assert.  I think there's a `;` missing:

```
      assert(s_a_dis[H_PORT] == FALSE) ***;***
      assert($countones({~s_a_dis[7:4] & bif_x8,~s_a_dis[3:0]}) > 1) else
```

solomatnikov, 2025-07-23:
Verilator doesn't handle correctly even simpler case:

```
  if ($test$plusargs("c"))
  assert(p_r_s[r_p].randomize() with {
                                                   if (t_p == topo_n_of_u) {
                                                      ...
                                                   } else {
                                                      ...
                                                   }                                                  
                                                   ...
                                                   });
  else    
  assert(p_r_s[r_p].randomize() with {
                                                   if (t_p == topo_n_of_u) {
                                                      ...
                                                   } else {
                                                      ...
                                                   }                                                  
                                                   ...
                                                   });
```

Note `;` after `assert`.

closed 2025-12-17 by wsnyder